Xilinx/XRT #4268: Question: support for OpenCL popcount ?

boxerab, 2020-10-19:
This is a 1.2 feature

https://manpages.debian.org/unstable/opencl-1.2-man-doc/popcount.3clc.en.html

Is it present in XRT ? If not, is there a library I can use to perform efficient popcount on Alveo boards ?

Thanks!

uday610, 2020-11-03:
Are you talking about OpenCl kernel code? Can you please confirm. 
Please show the usage with some sample code.

boxerab, 2020-11-03:
Thanks, yes, this is for kernel code.

uday610, 2020-11-05:
@boxerab , you can ask this in general Vitis support forum as XRT does not involve with the kernel support.  But in general, if you see Vitis examples, C Kernels are much more used and preferred for kernel coding. With C kernel you can get all functionalities and there are also several libraries.

boxerab, 2020-11-05:
Thanks, @uday610 but I am a bit confused. By kernel code, I mean code I can put into an OpenCL kernel:

```
const char *KernelSource = "\n" \
"__kernel void pop_square(                                              \n" \
"   __global uint* input,        	                                    \n" \
"   __global uint* output,  		                                    \n" \
"   const unsigned uint count)                                          \n" \
"{                                                                      \n" \
"   int i = get_global_id(0);                                           \n" \
"   uint pop = popcount(input[i]);                                    	\n" \
"   if(i < count)                                                       \n" \
"       output[i] = input[i] * input[i];                                \n" \
"}                                                                      \n" \
"\n";
```

Isn't this handled by XRT ?

boxerab, 2020-11-05:
I see now, I will ask about this on Vitis support forum.

closed 2020-11-05 by boxerab