Xilinx/XRT #4326: U250 validate

siniGAL, 2020-10-27:
Hi,

community.

I am setting up a development environment. Installing packages is fine.
At the stage of validation, the check does not end, it goes on forever.
I am attaching the dmesg log files and the output of other commands.

/opt/xilinx/xrt/bin/xbutil validate
INFO: Found 1 cards

INFO: Validating card[0]: xilinx_u250_xdma_201830_2
INFO: == Starting Kernel version check:
INFO: == Kernel version check PASSED
INFO: == Starting AUX power connector check:
INFO: == AUX power connector check PASSED
INFO: == Starting Power warnning check:
INFO: == Power warnning check PASSED
INFO: == Starting PCIE link check:
INFO: == PCIE link check PASSED
INFO: == Starting SC firmware version check:
INFO: == SC firmware version check PASSED
INFO: == Starting verify kernel test:
.................................... **it doesn't end**

1. lspci -vd 10ee:

03:00.0 Processing accelerators: Xilinx Corporation Device 5004
Subsystem: Xilinx Corporation Device 000e
Flags: bus master, fast devsel, latency 0
Memory at a2000000 (64-bit, prefetchable) [size=32M]
Memory at a4000000 (64-bit, prefetchable) [size=128K]
Capabilities: [40] Power Management version 3
Capabilities: [60] MSI-X: Enable+ Count=33 Masked-
Capabilities: [70] Express Endpoint, MSI 00
Capabilities: [100] Advanced Error Reporting
Capabilities: [1c0] #19
Capabilities: [400] Access Control Services
Capabilities: [410] #15
Kernel driver in use: xclmgmt
Kernel modules: xclmgmt

03:00.1 Processing accelerators: Xilinx Corporation Device 5005
Subsystem: Xilinx Corporation Device 000e
Flags: bus master, fast devsel, latency 0, IRQ 16
Memory at a0000000 (64-bit, prefetchable) [size=32M]
Memory at a4020000 (64-bit, prefetchable) [size=64K]
Memory at 90000000 (64-bit, prefetchable) [size=256M]
Capabilities: [40] Power Management version 3
Capabilities: [60] MSI-X: Enable+ Count=33 Masked-
Capabilities: [70] Express Endpoint, MSI 00
Capabilities: [100] Advanced Error Reporting
Capabilities: [400] Access Control Services
Capabilities: [410] #15
Kernel driver in use: xocl
Kernel modules: xocl

2. xbmgmt flash --scan

Card [0000:03:00.0]
Card type: u250
Flash type: SPI
Flashable partition running on FPGA:
xilinx_u250_xdma_201830_2,[ID=0x5d14fbe6],[SC=4.2.0]
Flashable partitions installed in system:
xilinx_u250_xdma_201830_2,[ID=0x5d14fbe6],[SC=4.2.0]

3. cat /etc/*-release

CentOS Linux release 8.2.2004 (Core)
NAME="CentOS Linux"
VERSION="8 (Core)"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="8"
PLATFORM_ID="platform:el8"
PRETTY_NAME="CentOS Linux 8 (Core)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:8"
HOME_URL="https://www.centos.org/"
BUG_REPORT_URL="https://bugs.centos.org/"

CENTOS_MANTISBT_PROJECT="CentOS-8"
CENTOS_MANTISBT_PROJECT_VERSION="8"
REDHAT_SUPPORT_PRODUCT="centos"
REDHAT_SUPPORT_PRODUCT_VERSION="8"

CentOS Linux release 8.2.2004 (Core)
CentOS Linux release 8.2.2004 (Core)

uday610, 2020-11-03:
Can you provide dmesg. 
Which XRT version you are using? Are you building XRT by git clone?

siniGAL, 2020-11-03:
Sorry, I forgot to answer. The problem was solved. I was building the XRT package from git. I saw a new commit and decided to try building the package again, after that the problem was solved.

closed 2020-11-05 by uday610