Xilinx/XRT #4466: Ubuntu 20.04 and 20.10 Vivado and XRT running well but not supported via installer.

Zaryob, 2020-11-15:


keryell, 2020-11-16:
After trying various ways, we reached the recipe https://github.com/triSYCL/sycl/blob/sycl/unified/master/sycl/doc/GettingStartedXilinxFPGA.md#install-vitis-20201 that worked for us for Ubuntu 20.04 & 2020.1. We upgraded the machine to 20.10 and it is still working.
Since we do not have tried 2020.2 yet with the installer on 20.10 from scratch I cannot be sure yet, but as we are skipping the non-working Java graphics part, it might work for you too...

stsoe, 2021-07-15:
Not an XRT issue.

closed 2021-07-15 by stsoe

keryell, 2021-07-15:
I guess there is a hierarchy of issues.
Working with a work-around while not supported is still great compared to not working at all.
Ping on https://github.com/Xilinx/XRT/issues/5168...