Xilinx/XRT #4487: libqdma_init() fails with DEBUG_SPINLOCK enabled

mclayton1000, 2020-11-19:
Hi all,

on Linux with DEBUG_SPINLOCK option enabled i get:

**[  250.590576] xocl:libqdma_init: dma req. opaque data size too big 136 > 128.**

This happens in **libqdma_init()** and is due to the fact that the structure **qdma_sgt_req_cb**
has a field of type wait_queue_head_t which is a structure that contains a field of type spinlock_t.
spinlock_t size depends on DEBUG_SPINLOCK configuration option.

The correct thing to do should be to turn the field **wq** of structure **qdma_sgt_req_cb** into a pointer
to avoid this issue.

Best regards.

mclayton1000, 2020-11-19:
Hi all,

if one change the **wq** type to pointer the module loading go ahead but stop on **libqdma4**:

**[ 1720.255593] xocl:libqdma4_init: dma req. opaque data size too big 128 > 80.**

The two errors look similar.

uday610, 2021-08-05:
@houlz0507 , this is old QDMA related issue. Is this still relevant? Let me know or close this.

houlz0507, 2021-08-05:
@uday610 , I do not know this. @karenx-xilinx  may know the detail.

uday610, 2021-08-05:
@karenx-xilinx , is this QDMA issue still relevant? Has this been fixed already or do we have internal story/cr to track this? Let me know or we can close this...

uday610, 2021-08-17:
As per @karenx-xilinx , this issue already raised in github QDMA driver . This is not a high priority item. Hence closing

closed 2021-08-17 by uday610