Xilinx/XRT #4497: Wrong xmaapi.h header copied to includes folder of XRT by the builder

gbonneau-hardent, 2020-11-25:
Once the build started with the script build.sh is finished, I have found that the Legacy XMA xmaapi.h header was copied to the includes folder of ../opt/xilinx/xrt/include/lib. This cause an issue for building an application that relies on this header since the XmaSingleton structure has changed with the new XMA 2 API. I am using branch 2020.1

sarab96, 2020-11-29:
XMA legacy headers are in /opt/xilinx/xrt/include (to be removed in near future)
XMA2 headers are in /opt/xilinx/xrt/include/xma2

gbonneau-hardent, 2020-11-29:
Indeed! Thanks for the head-up. However, this is somewhat confusing because the compile path of xmaapi.cpp for example uses the path "lib/xmaapi.h" with the correct version 2 include headers. Yet the build installs the include files of the legacy XMA within a folder with the same name lib. But this is probably for compatibility issues. Great then we can close this issue. - Thanks.

closed 2020-11-29 by gbonneau-hardent