Xilinx/XRT #4502: master failing tests

boxerab, 2020-11-26:
build command:  `./build.sh -opt`

```
Test project /home/src/XRT/build/Release
    Start 2: xbutil
    Start 3: xbutil2
    Start 4: xbmgmt2
    Start 1: xbmgmt
1/4 Test #2: xbutil ...........................***Failed    0.00 sec
/home/src/XRT/build/Release/runtime_src/core/pcie/tools/xbutil/xbutil: symbol lookup error: /home/src/XRT/build/Release/runtime_src/core/pcie/tools/xbutil/xbutil: undefined symbol: _ZN6pcidev16check_p2p_configERKSt10shared_ptrINS_10pci_deviceEERNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE

2/4 Test #3: xbutil2 ..........................***Failed    0.00 sec
/home/src/XRT/build/Release/runtime_src/core/tools/xbutil2/xbutil2: symbol lookup error: /home/src/XRT/build/Release/runtime_src/core/tools/xbutil2/xbutil2: undefined symbol: _ZN3xrt5errorC1ERKNS_6deviceE13xrtErrorClass

3/4 Test #1: xbmgmt ...........................   Passed    0.00 sec
4/4 Test #4: xbmgmt2 ..........................***Failed    0.00 sec
/home/src/XRT/build/Release/runtime_src/core/tools/xbmgmt2/xbmgmt2: symbol lookup error: /home/src/XRT/build/Release/runtime_src/core/tools/xbmgmt2/xbmgmt2: undefined symbol: _ZN3xrt5errorC1ERKNS_6deviceE13xrtErrorClass

25% tests passed, 3 tests failed out of 4

Total Test time (real) =   0.01 sec

The following tests FAILED:
	  2 - xbutil (Failed)
	  3 - xbutil2 (Failed)
	  4 - xbmgmt2 (Failed)
Errors while running CTest
```

stsoe, 2020-11-28:
hi @boxerab As you can imagine, master doesn't fail for us when we build XRT.  Can you provide some more context please?

boxerab, 2020-11-28:
Thanks, @stsoe . I pulled the code just now and there are no errors.

closed 2020-11-28 by boxerab