Xilinx/XRT #4504: CMake Error: install(EXPORT "xrt-targets" ...) includes target "xilinxopencl" more than once in the export set.

mclayton1000, 2020-11-27:
Hello everyone,

recent commits causes the XRT not to build anymore. Here is the error:

**-- Preparing XRT find_package
-- Configuring done
CMake Error: install(EXPORT "xrt-targets" ...) includes target "xilinxopencl" more than once in the export set.
-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.**

The offending commits are:
c2ce13de41d507edd37edfb7e101b83c2aa78224  / Add support for cmake's find_package on linux
fb91637466c62f0687caa43a69bee4ec445925fe   / Create relocatable export files

stsoe, 2020-11-27:
Hello @mclayton1000 ,  Please provide more context.  XRT builds on all the platforms we support officially.

mclayton1000, 2020-11-27:
Hi @stsoe ,

thank you for the reply and support.

I found that the problem is related to the following files:
src/runtime_src/core/common/CMakeLists.txt
src/runtime_src/core/pcie/emulation/cpu_em/CMakeLists.txt
src/runtime_src/core/pcie/emulation/hw_em/CMakeLists.txt
src/runtime_src/core/pcie/linux/CMakeLists.txt
src/runtime_src/xocl/CMakeLists.txt
src/runtime_src/xrt/CMakeLists.txt

In particular the problem is that in each of those files there is cmake code like:

install(TARGETS <something>
  EXPORT xrt-targets
  LIBRARY DESTINATION ${XRT_INSTALL_LIB_DIR} ${XRT_NAMELINK_SKIP}
  RUNTIME DESTINATION ${XRT_INSTALL_BIN_DIR}
)
install(TARGETS <something>
 EXPORT xrt-targets
 LIBRARY DESTINATION ${XRT_INSTALL_LIB_DIR} ${XRT_NAMELINK_SKIP}
 RUNTIME DESTINATION ${XRT_INSTALL_BIN_DIR}
)

where <something> is:
xrt_coreutil
xrt_swemu
xrt_hwemu
xrt_core
xilinxopencl
xrt++

Commenting out in each file one of the two cmake install command solve the issue.
Is it possible to merge that commands?

I'm using **cmake 3.19.1.**

stsoe, 2020-11-27:
Thanks.  Which OS are you building on?

I will fix this once I am able to reproduce, one of the EXPORTs should be removed (I think the one from the COMPONENT release) but I need to double check.

mclayton1000, 2020-11-27:
Hi @stsoe ,

I'm on **Slackware-current** but I saw that the same cmake version is also used on Arch Linux.

Please, ask me if you need any help.

Thank you for your support.

stsoe, 2020-11-27:
Hi @mclayton1000.  It would be great if you could try remove the EXPORT from the second INSTALL command (the one with XRT_DEV_COMPONENT) in each of the make files you mentioned.    I will do the same on my end and check if I can still use find_package.  Thanks!

mclayton1000, 2020-11-27:
Hi @stsoe ,

removing the second INSTALL commands solved the issue.

Thank you for the support, best regards.

stsoe, 2020-11-27:
Hi @mclayton1000.  Just to clarify, you removed the EXPORT from the second INSTALL command, not the command itself, right?
I will commit this change as it also worked with find_package on my side.   Thank you for pointing this out.

mclayton1000, 2020-11-27:
Hi @stsoe ,

I confirm you that I only removed the EXPORT option from the second install command.

Best regards.

PR #4505 (2020-11-27): Remove duplicate EXPORT option
Some versions of CMake or OS'es complain about target included more
than once in EXPORT.

Fixes #4504
commit 9cfed4e5 (2020-11-27): Remove duplicate EXPORT option

Some versions of CMake or OS'es complain about target inclide more
than once in EXPORT.

Fixes #4504
closed 2020-11-27 by stsoe