Xilinx/XRT #4507: XRT Master:  exception during program exit

boxerab, 2020-11-28:
```
terminate called after throwing an instance of 'xrt_core::error'
  what():  failed to unmap BO
```

Here is the stack trace:

```
__GI_raise() at raise.c:51 0x7ffff7229fb7	
__GI_abort() at abort.c:79 0x7ffff722b921	
0x7ffff788e257	
0x7ffff7899606	
0x7ffff7898649	
__gxx_personality_v0() at 0x7ffff7898ff1	
0x7ffff75ebeff	
_Unwind_RaiseException() at 0x7ffff75ec581	
__cxa_throw() at 0x7ffff78998f8	
xrt_core::shim<xrt_core::device_pcie>::unmap_bo() at 0x7ffff2dfd083	
std::_Sp_counted_ptr_inplace<xrt::buffer_kbuf, std::allocator<xrt::buffer_kbuf>, (__gnu_cxx::_Lock_policy)2>::_M_dispose at 0x7ffff643cec9	
xocl::platform::~platform() at 0x7ffff69ee4ea	
std::shared_ptr<xocl::platform>::~shared_ptr() at 0x7ffff69eed7a	
__run_exit_handlers() at exit.c:108 0x7ffff722e161	
__GI_exit() at exit.c:139 0x7ffff722e25a	
__libc_start_main() at libc-start.c:344 0x7ffff720cbfe	
```

stsoe, 2020-12-08:
Hi @boxerab  

I think the host code is the reason for the crash. I spend some time debugging this, andit looks like the reference counts on the OpenCL objects are not managed correctly.  For example:

- `DeviceManagerOCL` calls `clReleaseCommandQueue` for the one reference it manages on the command queue (this is good).
- However, wide_vadd_main.cpp passes the ownership of `dev->queue` to `cl::CommandQueue q(dev->queue)`.
- When `q` goes out of scope, it too will release the command queue

This is just one example, as far as I can tell there are multiple other problems in the host code related reference counting of object management.

boxerab, 2020-12-08:
Thanks, @stsoe , you are right, and sorry for the noise.

closed 2020-12-08 by boxerab