Xilinx/XRT #4516: XMA CU Kernel command start fail when Kernel is not a dataflow kernel

gbonneau-hardent, 2020-12-01:
This was tested with the HW Emulator. If the HW emulation behavior is consistent then the issue is likely to be present when HW emulation is not used. The problem seems to be related to the wrong assumption from the function xma_plg_schedule_work_item() that an ERT packet must be created with all the 3 extra data masks. This seems to be related to the specification of the ERT Packet command. However, if you don't use a kernel with dataflow the ERT is disabled and the Scheduler doesn't use polling. Later when the command is submitted all the 3 extra masks are wrongly set up in the register space command. Then the kernel fails since it doesn't recognize the command.
The issue can easily be reproduced by using an RTL Kernel rather than an HLS C/C++ Kernel. I have been able to successfully run the RTL kernel (in HW emulation) with XMA by removing the extra data mask from function xma_plg_schedule_work_item().

sarab96, 2020-12-09:
Good check on HW emulation. We will look into it. Note that XMA does supports HW emulation but not software emulation

sarab96, 2021-07-22:
This is HW emulation only issue where cu_mask entries are not handled properly