Xilinx/XRT #4524: Build of XRT of releases branch 2020.2 fails on Ubuntu 18.04 (with boost 1.65.1)

gbonneau-hardent, 2020-12-03:
In file included from /usr/include/boost/process/detail/posix/executor.hpp:15:0,
                 from /usr/include/boost/process/detail/execute_impl.hpp:24,
                 from /usr/include/boost/process/child.hpp:22,
                 from /usr/include/boost/process/async_system.hpp:22,
                 from /usr/include/boost/process.hpp:24,
                 from /home/guy/git/XRT/src/runtime_src/core/tools/common/Process.cpp:45:
/usr/include/boost/process/pipe.hpp: In instantiation of ‘boost::process::basic_pipebuf<CharT, Traits>* boost::process::basic_ipstream<CharT, Traits>::rdbuf() const [with CharT = char; Traits = std::char_traits<char>]’:
/home/guy/git/XRT/src/runtime_src/core/tools/common/Process.cpp:219:32:   required from here
/usr/include/boost/process/pipe.hpp:265:57: error: cannot convert ‘const boost::process::basic_pipebuf<char>’ to ‘boost::process::basic_pipebuf<char>*’ in return
     basic_pipebuf<CharT, Traits>* rdbuf() const {return _buf;};

gbonneau-hardent, 2020-12-03:
Note: The workaround is to cherry pick: commit ID 129a11a CR-1079673 XRT build failure with cmake 3.17 and boost 1.65.1 (#4461)

stsoe, 2020-12-11:
Hi @gbonneau-hardent  The mentioned cherry pick was merged to 2020.2 to resolve your build issue.

closed 2020-12-11 by stsoe