Xilinx/XRT #4525: BUG:

mclayton1000, 2020-12-03:
Hello everyone, hi @karenx-xilinx , hi @houlz0507 , hi @sonals , hi @akum-xilinx , hi @pranjalv-xilinx 
 
with mutex debugging enabled on Linux kernel I get the following on **xclmgmt.ko** module loading:

**[  169.414728] xocl 0000:01:00.1:  ffff8a09f4d408b0 __xocl_subdev_create: Created subdev dma.xdma inst 5242880 level 0
[  169.414731] xocl 0000:01:00.1:  ffff8a09f4d408b0 __xocl_subdev_create: subdev dma.xdma.u.5242880 inst 5242880 is active
[  169.414735] xocl 0000:01:00.1:  ffff8a09f4d408b0 __xocl_subdev_construct: creating subdev mailbox.u multi 0 level 0
[  169.414739] xocl 0000:01:00.1:  ffff8a09f4d408b0 __xocl_subdev_construct: resource [mem 0xf0200000-0xf020002f]
[  169.414764] xocl 0000:01:00.1: mailbox.u.9437184 ffff8a09f4320c10 mailbox_start: Starting Mailbox channels
[  169.415612] BUG: sleeping function called from invalid context at /opt/repos/linux-5.4.78/kernel/locking/mutex.c:281
[  169.415617] in_atomic(): 1, irqs_disabled(): 1, non_block: 0, pid: 1971, name: insmod
[  169.415620] CPU: 0 PID: 1971 Comm: insmod Tainted: G           O      5.4.78 #5
[  169.415623] Hardware name: Gigabyte Technology Co., Ltd. B250M-D3H/B250M-D3H-CF, BIOS F9b 01/10/2018
[  169.415625] Call Trace:
[  169.415630]  dump_stack+0x57/0x75
[  169.415634]  ___might_sleep+0xe9/0x110
[  169.415636]  __might_sleep+0x45/0x80
[  169.415666]  ? user_intr_register+0x310/0x310 [xocl]
[  169.415670]  mutex_lock+0x1b/0x50
[  169.415696]  xdev_find_by_pdev+0x16/0x70 [xocl]
[  169.415720]  xdma_user_isr_register+0x37/0xf0 [xocl]
[  169.415740]  user_intr_register+0xca/0x310 [xocl]
[  169.415761]  ? chan_timer+0x40/0x40 [xocl]
[  169.415789]  userpf_intr_register+0x45/0x100 [xocl]
[  169.415810]  mailbox_enable_intr_mode+0xba/0x250 [xocl]
[  169.415830]  mailbox_start+0x1a7/0x3c0 [xocl]
[  169.415848]  mailbox_probe+0xd6/0x220 [xocl]
[  169.415852]  platform_drv_probe+0x37/0x90
[  169.415854]  ? driver_sysfs_add+0x71/0xa0
[  169.415856]  really_probe+0x1bf/0x320
[  169.415858]  driver_probe_device+0x50/0xf0
[  169.415860]  __device_attach_driver+0x6b/0x90
....**

It seems that someone is trying to get a mutex in kernel atomic context.

Please, can you fix this to get more robust code?

Thank you, best regards.

houlz0507, 2020-12-08:
@mclayton1000, Thanks for finding this. We will fix it.

mamin506, 2020-12-09:
@mclayton1000, fix is merged. Please verify this at your end as well. Thanks for reporting. I would close this issue now.

closed 2020-12-09 by mamin506