Xilinx/XRT #4541: Disable run-time reconfiguration

zohourih, 2020-12-08:
Is there a way (environmental variable or function flag in host code) to disable run-time reconfiguration of the FPGA in Vitis? The goal is to program the FPGA manually using "xbutil program" and then running the host code without the "xclbin" file being present. The run-time somehow automatically detects when the FPGA is already configured with the bitstream that is provided (e.g. if you run the same host code more than once), and does not reconfigure the FPGA with the same bitstream more than once, but this requires the bitstream to be available and loaded. I am hoping there was a way to achieve this without the bitstream being present.

stsoe, 2020-12-08:
Hi @zohourih  Multi-process control of kernels is not supported without all processes having access to the xclbin.  This is because in order to control the kernels, a process needs to obtain meta-data which is available only in the xclbin, not in the bitstream that is loaded on the device.   The good news is that only one process will be loading the bitstream, once loaded, subsequent loading of the same bitstream by other processes will be a no-op. 

Having said that, we are working on caching the meta data from the xclbin such that it is available for use by processes that do not have access to the xclbin.    This is something we will support in a later release.

Please close this issue if I have answered your question.  Thanks!

zohourih, 2020-12-08:
@stsoe The issues is not related to multi-processing, it is a much simpler matter. I am trying to pre-program the FPGA using `xbutil program`, then delete the bitstream, and then run the host code without needing the bitstream since the FPGA is already programmed with it. I think I have found a work-around for this, so I will close this issue for now.

zohourih, 2020-12-08:
Looks like I have to re-open the issue. What I trying to do right now is programming the FPGA with the `xclbin` file, stripping away the "BITSTREAM" section from the `xclbin` file and saving the resulting file in a new file, and then running the host code using the new `xclbin` which does _not_ include the BITSTREAM section, hoping that XRT is smart enough to avoid run-time reconfiguration. However, the host code fails with the following error in this case:

```
Found Platform
Platform Name: Xilinx
INFO: Reading ./test.xclbin
Loading: './test.xclbin'
XRT build version: 2.7.766
Build hash: 19bc791a7d9b54ecc23644649c3ea2c2ea31821c
Build date: 2020-08-17 16:52:05
Git branch: 2020.1_PU1
PID: 32004
UID: 1001
[Tue Dec  8 22:23:47 2020 GMT]
HOST: Edge-tower
EXE: /home/projects/host
[XRT] ERROR: See dmesg log for details. err=-22
[XRT] ERROR: Failed to load xclbin.
/home/projects/common/xilinx/xcl2/xcl2.cpp:132 Error calling *program = cl::Program(*context, {device}, bins, NULL, &err), error code is: -44
```

How can I run the host code with the stripped-down `xclbin`?

stsoe, 2020-12-09:
@zohourih As I mentioned even without stripping the bitstream, XRT will not re-program the device if it is already programmed with the same bitstream.   However, stripping the bitstream, to reduce file size for subsequent reading from disk, is something we have done in the past and I just verified that it does indeed work.    However the command to strip the bitstream while pretending the xclbin is the same is this one:
`xclbinutil --skip-uuid-insertion --remove-section BITSTREAM ...`

zohourih, 2020-12-09:
@stsoe Thank you for the tip, adding `--skip-uuid-insertion` to the command line solves the issue for me.

closed 2020-12-09 by zohourih