Xilinx/XRT #4564: WARNING: Incomplete CU profile trace detected. Timeline trace will have approximate CU End

doonny, 2020-12-15:
I am test my design with Vitis 2020.1 on a Ubuntu 18.04 machine with the U50 boards.
The XRT generates the following warning when performing hw emulation:
```
WARNING: Incomplete CU profile trace detected. Timeline trace will have approximate CU End
```
The xrt.ini setting is as follows:

[Debug]
profile=true
#power_profile=true
timeline_trace=true
data_transfer_trace=fine
#stall_trace=all
trace_buffer_size=512M
[Emulation]
debug_mode=batch

uday610, 2021-02-08:
You may check if you are getting the same message in hardware flow as well. 
One of the reasons is that your host code execution finished but the internal simulation is still running . Perhaps any event in the host code did not finish. Can you make sure you have cleaned all OpenCL objects in the host code? Thanks

anupcs-xlnx, 2021-07-07:
Please see previous comment from @uday610

closed 2021-07-07 by anupcs-xlnx