Xilinx/XRT #4581: host-kernel streaming causes Linux memory leak

Blaok, 2020-12-21:
Hi,

It seems that host-kernel streaming is causing memory leak in the Linux kernel. The active memory usage (`grep 'Active(anon)' /proc/meminfo`) keeps increasing as I repeatedly run [this example from Xilinx/Vitis_Accel_Examples](https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/host/streaming_host_bandwidth). Each execution is successful and exits cleanly ([after I disable the non-blocking part](https://github.com/Xilinx/Vitis_Accel_Examples/issues/32)), so the memory leak can only be attributed to the Linux kernel driver, which is part of XRT. The same memory leak happens with programs that I wrote myself, so I don't think it is a bug in the example code.

Output from `xbutil scan`:

```plaintext
INFO: Found total 4 card(s), 4 are usable
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
System Configuration
OS name:	Linux
Release:	4.15.0-128-generic
Version:	#131-Ubuntu SMP Wed Dec 9 06:57:35 UTC 2020
Machine:	x86_64
Model:		PowerEdge R740
CPU cores:	32
Memory:		95127 MB
Glibc:		2.27
Distribution:	Ubuntu 18.04.4 LTS
Now:		Mon Dec 21 06:50:32 2020 GMT
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
XRT Information
Version:	2.8.726
Git Hash:	7c93966ead2dec777b92bdc379893f22b5bd561e
Git Branch:	2020.2
Build Date:	2020-11-11 20:29:19
XOCL:		2.8.726,7c93966ead2dec777b92bdc379893f22b5bd561e
XCLMGMT:	2.8.726,7c93966ead2dec777b92bdc379893f22b5bd561e
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 [0] 0000:d8:00.1 xilinx_u280_xdma_201920_3(ID=0x5e278820) user(inst=131)
 [1] 0000:af:00.1 xilinx_u250_xdma_201830_2(ID=0x5d14fbe6) user(inst=130)
 [2] 0000:5e:00.1 xilinx_u50_gen3x16_xdma_201920_3 user(inst=129)
 [3] 0000:3b:00.1 xilinx_u200_qdma_201920_1(ID=0x5dccb0ca) user(inst=128)

```

Thank you!

maxzhen, 2020-12-21:
@mamin506 , could you take a look and see if this is kds issue?

mamin506, 2020-12-22:
@maxzhen It doesn't looks like kds issue. On my test, for each run, the Active(anon) memory increased about 2,300,000 kB.
I saw below error by dmesg.

[  514.786898] xocl 0000:05:00.1: dma.qdma.u.5242881 ffff9d6b3bf3f410 queue_rw: queue_rw ret 1073741824, kiocb 0x          (null).
[  514.787699] xocl 0000:05:00.1: dma.qdma.u.5242881 ffff9d6b3bf3f410 queue_rw: queue_rw ret 1073741824, kiocb 0x          (null).
[  522.021716] xocl 0000:05:00.1: dma.qdma.u.5242881 ffff9d6b3bf3f410 queue_flush: Release Queue 0x210
[  522.032129] xocl:qdma_queue_stop: qdma05001-ST-C2H-16 16 stopped.
[  522.039246] xocl:error_intr_handler: Error IRQ fired on PF#1: index=8, vector=107
[  522.047682] xocl:err_stat_handler: glbl_err[0x248] : 0x100
[  522.053012] xocl:err_stat_handler:   err_c2h_st detected
[  522.059279] xocl:err_stat_handler: c2h_err[0xaf0] : 0x1000
[  522.063297] xocl:err_stat_handler:   cmpt_inv_q_err detected

-----------------------------------------------------------------------------------------
If I run xbutil validate on this QDMA shell, there is no memory leak and error in dmesg.
@houlz0507 Could you please take a look at this case? I can provide setup to reproduce it.

houlz0507, 2021-01-05:
fixing it by https://github.com/Xilinx/XRT/pull/4595

Blaok, 2021-01-05:
I built from 0a1d003924781858856f38060e9b0d103aa26384 and it works. Thank you!

closed 2021-01-05 by Blaok