Xilinx/XRT #4585: xma_xclbin_file_open() doesn't check for file existence.

gbonneau-hardent, 2020-12-26:
When a call is made to open the xclbin file if it doesn't exist -1 is returned from infile.tellg() causing a bad alloc memory and a throw. 

Can I suggest that a std::vector size of zero should be returned to provide a mean to check that the call to xma_xclbin_file_open is unsuccessful and provide a mean to return from xma_initialize(...) with an error. As much as possible all internal call paths should try to have xma_intialize() return with a failing error or a success. Not all programs calling xma_intialize() are C++ and can surround the calling with try...catch... FFMpeg for example.

keryell, 2020-12-27:
But if it returns an `.empty()` `std::vector` you need *also* a C++ program anyway...

gbonneau-hardent, 2020-12-27:
Not if you use the XMA API. xma_xclbin_file_open() is called from the C++ XMA layer and is not an external XMA API. Thus it is an issue.

sarab96, 2021-07-22:
No change at this time

closed 2021-07-22 by sarab96