Xilinx/XRT #4645: XRT conflict with Vitis during build on Ubuntu 20.04

michael-barrow, 2021-01-21:
I am trying to follow the https://xilinx.github.io/XRT/master/html/build.html build instructions, but these conflict with the method for setting XILINX_VITIS via the supplied settings64.sh for the Vitis tool.

Vitis ships with cmake 3.2.x but this library requires cmake > 3.5.x

the workaround was to modify build.sh such that CMAKE=/usr/bin/cmake but this is not ideal for other distros

keryell, 2021-01-21:
I looked at https://xilinx.github.io/XRT/master/html/build.html and there is nothing about `settings64.sh` there.

Sourcing `settings64.sh` is used to set up the environment to run Vitis. Unfortunately it is so intrusive that if breaks almost everything else. So, my advice is: do not use this in a shell where your plan is not to use only Vitis.

You should not source Vitis configuration file by default in your shell configuration (`.bashrc` or equivalent) or it will break deeply your system.

An easy way is to use 1 terminal for Vitis where you `source settings64.sh` and other terminal to do anything else.

stsoe, 2021-01-24:
I am closing this issue based on @keryell feedback

closed 2021-01-24 by stsoe