Xilinx/XRT #4696: Arm服务器上Xilinx XRT的编译问题

DaveAimee, 2021-02-05:
1. 编译报错

[image]
2. 编译时是否需要安装Xilinx Vitis Software Platform
目前已经尝试下载了Xilinx_Unified_2020.2_1118_1232_Lin64.bin，但是在安装的时候提示：
[image]

maxzhen, 2021-02-05:
@DaveAimee please let us know the Linux kernel version (uname -r).

DaveAimee, 2021-02-19:
My "uname -r" shows 4.14.0-115.el7a.0.1.aarch64, does xrt hava specific version requirements?

DaveAimee, 2021-02-19:
Checked once again,found I forgot to run './build.sh clean' to clean some stale configs. My compile problem has been solved, Thanks!

closed 2021-02-19 by DaveAimee