Xilinx/XRT #4734: XILINX_VITIS documentation

vmayoral, 2021-02-16:
After extending the simple example provided at [the tutorials](https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/Part4.md) with other setups (involving artifacts generated and placed in different folders) I noticed It'd be nice to get **more documentation** on the behavior/importance/meaning of the `XILINX_VITIS` variable.

https://www.xilinx.com/html_docs/xilinx2020_1/vitis_doc/runningapplication.html doesn't really mention it. Happy to provide a few lines documenting my experience if you point me out "where" it should be done.

uday610, 2021-02-18:
@vmayoral , can you please elaborate a bit what you thought would be nice inside the vitis_doc running application section regarding XILINX_VITIS variable?

vmayoral, 2021-02-19:
@uday610, I believe that beyond XRT, `XILINX_VITIS` relates to Vitis source. I don't have access to this so my perspective is terribly biased and *purely based on my experimentation*, which isn't much for now.

Note the following may **NOT** be accurate at all (that's why I was hoping to get *official* documentation in place). So far, my notes indicate the following:
- `XILINX_VITIS` variable isn't required to operate in embedded platforms or emulation
- `XILINX_VITIS` variable allows to define a path where-to-look-for accelerated kernels and related artifacts *however*, full paths still need to provided when interfacing with OpenCL.

I'm sure there's more to it. Hope this helps.

uday610, 2021-03-08:
@vmayoral , I dont think that is accurate. XILINX_VITIS path is required for emulation. To make things seamless for the user we always recommend to set XILINX_VITIS path, if you see the top of the page 

https://www.xilinx.com/html_docs/xilinx2020_1/vitis_doc/runningapplication.html

"Set up the command shell or window as described in Setting up the Vitis Environment prior to running the builds."  this point to another page (https://www.xilinx.com/html_docs/xilinx2020_1/vitis_doc/settingupvitisenvironment.html#zks1565446519267) which clearly says to source two .sh files namely ``<Vitis_install_path>/settings64.sh`` and ``/opt/xilinx/xrt/setup.sh` ` .. So we think as long as the above two paths are set the user should not have any issue whether they are targeting Alveo card, or embedded platform, or even emulation flow

vmayoral, 2021-03-09:
> @vmayoral , I dont think that is accurate. XILINX_VITIS path is required for emulation. To make things seamless for the user we always recommend to set XILINX_VITIS path, if you see the top of the page

@uday610 thanks for answering but I disagree. Everything above is meant on the "embedded" environment. 
Please read again my comments with that perspective in mind. Of course in the workstation those variables are set if you follow the instructions. In the board **however**, such variable is recommended to be set but nowhere documented. If it is, point me out to where please.

vmayoral, 2021-05-07:
`XILINX_VITIS` has dissapeared from https://github.com/Xilinx/Vitis-Tutorials/blob/master/Getting_Started/Vitis/Part4.md. Has this variable been removed completely from tutorials, should we then close this ticket?

vmayoral, 2021-05-08:
> The $XILINX_VITIS variable is needed to point to the installation of Xilinx Vitis tools and libraries. It is not needed to run your application on the hw platform. However, it is needed to run your application in the QEMU environment for emulation, as it is used to launch and configure the QEMU environment.

@randyh62 thanks for the answer. Do you mean `XILINX_VITIS` needs to set in the workstation to launch emulations? If that's the case, I agree and let's close this. Otherwise, if what you mean is that we need that variable inside of QEMU emulations, then that doesn't match my experience so far. I can perfectly run emulation without `XILINX_VITIS` (either simulating or emulating PL, `sw_emu` or `hw_emu`) and still get perfectly fine results without.

vmayoral, 2021-05-09:
Closing for now.

closed 2021-05-09 by vmayoral