Xilinx/XRT #4735: Using CMake modules instead of files, extensibility and composability

vmayoral, 2021-02-16:
**Context**: I'm engaged in a project making heavy use of acceleration and I'm considering to build on top of XRT as a submodule instead of reinventing the wheel and rewriting my own framework/runtime. 

---

Current CMake structure `include()` *files* instead of *modules*. While the difference could seem small, it actually isn't extensibility-wise and/or composability-wise. When using XRT as a submodule, all sorts of funny things happen when connecting and chaining it to the external build infrastructure (even if also CMake-based!).

I've documented this slightly in my extensions as follows:
```cmake
# NOTE: include function will only use CMAKE_MODULE_PATH
#  if the provided name is a "module" (not a file).
# See https://stackoverflow.com/a/59326925 for more details.

# NOTE 2: accordingly, all "include"-calls should be re-written
# and instead of receiving "files" (with a .cmake), should receive
# modules (essentially, without the .cmake). The following will cause
# chaos:
#  set(CMAKE_MODULE_PATH ${CMAKE_SOURCE_DIR}/src)
```

Addressing this could be done by replacing the `include()` calls and using *modules* instead of *files*. This way, extending outer build setups with `CMAKE_MODULE_PATH` to include XRT will be feasible.

Before committing to any contribution, let's see what's the reaction. Ping @stsoe @chienwei-lan @houlz0507 and other forks maintaining this.

*Side note*: I'm not sure who's "in charge" of maintaing the project but I'd be great to list this somewhere. If I missed this, apologies and please point me out to where it is.

stsoe, 2021-02-16:
Hi @vmayoral This sounds reasonable.  Is the only required change to replace` include(foo.cmake)` with `include(foo)`?

vmayoral, 2021-02-16:
> Hi @vmayoral This sounds reasonable. Is the only required change to replace include(foo.cmake) with include(foo)?

Hey @stsoe, yes, that'll do. Will need to happen though in quite a few files. Let me know if you want a PR for that.

stsoe, 2021-02-16:
Thanks @vmayoral, it would be great with a PR.

uday610, 2021-07-21:
no response, closing

closed 2021-07-21 by uday610