Xilinx/XRT #4736: Hardcoded paths in CMake setup, issues for extensibility and composability

vmayoral, 2021-02-16:
Current XRT codebase uses harcoded paths pointing to `/usr/src/*`, `/opt/xilinx/xrt/share/*` or `/usr/lib64/*`among others. While this is a reasonably design choice for stability of the runtime itself on a given set of targeted platforms, if planning to use it as a submodule, such hardcoded paths become a hurdle.

Are you willing to accept changes that either install files relative to `CMAKE_INSTALL_PREFIX` or allow for external build setups to do so (e.g. by means of using conditional conditions through variables which if defined will trigger the corresponding condition)?

Related to https://github.com/Xilinx/XRT/issues/4735

stsoe, 2021-02-16:
Hi @vmayoral.   I believe XRT is installing relative to CMAKE_INSTALL_PREFIX, but I do see CMAKE_INSTALL_PATH hard-coded.  Can you point specifically to other hard-coded paths that are a problem?

vmayoral, 2021-02-16:
>  Can you point specifically to other hard-coded paths that are a problem?

Sure:

- https://github.com/Xilinx/XRT/blob/master/src/CMake/dkms-aws.cmake#L6
- https://github.com/Xilinx/XRT/blob/master/src/CMake/dkms.cmake#L6
- https://github.com/Xilinx/XRT/blob/master/src/CMake/icd.cmake#L10
- etc

Something like the following would do for those of us aiming to extend it:

```cmake
if(<condition-set-externally, e.g. variable>)
  set(OCL_ICD_INSTALL_PREFIX "${CMAKE_INSTALL_PREFIX}/OpenCL/vendors")
else()
  set(OCL_ICD_INSTALL_PREFIX "/etc/OpenCL/vendors")
endif()  # ament_cmake_FOUND

```

stsoe, 2021-02-16:
Got it thanks.   I'd like the various needed prefixes consolidated in one place (e.g. src/CMakeLists.txt) and individual .cmake files to use the consolidated prefixes.   I see that src/CMakeLists.txt also blindly sets CMAKE_INSTALL_PREFIX so this along with the necessary prefixes could be guarded as you suggest in this one place, right?  @vmayoral you mind creating a PR that you have a verified works to your needs?  Thanks!

uday610, 2021-07-21:
no response, hence closing.

closed 2021-07-21 by uday610