Xilinx/XRT #4792: XRT build error - protobuf generated files version mismatch

rudolpjo, 2021-02-24:
I have a more thorough explanation of the issue on the Xilinx forum here: https://forums.xilinx.com/t5/AI-and-Vitis-AI/Xilinx-Run-Time-XRT-build-script-error-protobuf-version/td-p/1210669

The following errors after the initial one mentioned in the post are all related to the protobuf library in one way on another.

rudolpjo, 2021-03-01:
Can someone please take a look at this?

sonals, 2021-03-04:
XRT uses native libraries that ship with the Linux distribution. Can you use native protobuf that comes with Ubuntu? Install the following packages with **apt install** and remove your own build of protobuf:

```
libprotoc-dev 
protobuf-compiler
```

rudolpjo, 2021-03-04:
Thank you for responding to this issue, I tried what you suggested but got the same errors. Interestingly enough, after I removed the protobuf files from `/usr/include/google` and ran apt install, nothing was downloaded or updated. Got the message that all was up to date. Perhaps I am missing something with the location of the needed dependencies. 
When the issue first occurred, I was using the native Ubuntu 18.04 libraries, and there was a protobuf error regarding a missing file. I have tried searching for what version of protobuf ships with Ubuntu 18.04, but I have not been able to find that information out.

rudolpjo, 2021-03-05:
Solved: 
Check version with command
`protoc --version`
mine was 3.14
Go to `google/protoc/stubs/common.h` look for version number.
The version installed was `3015000`, 3.15
Remove 3.15, download and install 3.14 from github.

closed 2021-03-05 by rudolpjo