Xilinx/XRT #5168: Kernel driver: deprecated gem_vm_ops has been removed in Linux 5.11

j-stephan, 2021-04-28:
In Linux 5.4, several members of `struct drm_driver` have been deprecated in favour of their counterparts in `struct drm_gem_object_funcs`. In 5.11, they have been removed. This causes the DKMS compilation to fail on up-to-date Linux distributions, such as Ubuntu 21.04:

https://github.com/Xilinx/XRT/blob/05c9c23f38a111998a5adcb324fdbe681e9d9780/src/runtime_src/core/pcie/driver/linux/xocl/userpf/xocl_drm.c#L516

https://github.com/Xilinx/XRT/blob/05c9c23f38a111998a5adcb324fdbe681e9d9780/src/runtime_src/core/edge/drm/zocl/zocl_drv.c#L780

https://github.com/Xilinx/XRT/blob/05c9c23f38a111998a5adcb324fdbe681e9d9780/src/runtime_src/core/pcie/driver/linux/xocl/userpf/xocl_bo.c#L1231

Error message:

```
  CC [M]  /var/lib/dkms/xrt/2.11.0/build/driver/xocl/userpf/xocl_bo.o
/var/lib/dkms/xrt/2.11.0/build/driver/xocl/userpf/xocl_bo.c: In function ‘xocl_gem_prime_mmap’:
/var/lib/dkms/xrt/2.11.0/build/driver/xocl/userpf/xocl_bo.c:1231:39: error: ‘const struct drm_driver’ has no member named ‘gem_vm_ops’
 1231 |   vma->vm_ops = xobj->base.dev->driver->gem_vm_ops;
      |                                       ^~
```

keryell, 2021-04-30:
Ouch... I have just updated my laptop to Ubuntu 21.04... :-|

keryell, 2021-05-14:
For the Xilinx team, there is the machine `xsjsycl40` with Ubuntu 21.04 and kernel 5.11 to compile. There is some local storage in `/var/tmp` and `/storage`.
But anyway, since the new Linux API is available since Linux 5.4, any relatively recent Linux distribution has a kernel by default supporting the new API, such as Ubuntu 20.04 LTS.
This is blocking the upgrade of `xsjsycl41` with an Alveo U200 to Ubuntu 21.04...

keryell, 2021-05-26:
Problem also mentioned in https://forums.xilinx.com/t5/Vitis-Acceleration-SDAccel-SDSoC/Patch-XRT-install-for-use-with-newer-Linux-kernels/td-p/1244066
I just want to make sure we are not duplicating the effort if someone is motivated by fixing the issue. :-)
Thus, someone starting working on this could write something here to avoid mid-air work collision.

anadon, 2021-05-26:
Got it.  I'll see about a patch tonight, the cosmos permitting.

keryell, 2021-06-16:
> Got it. I'll see about a patch tonight, the cosmos permitting.

Any news from the cosmos permissions? :-)

anadon, 2021-06-16:
Well, I knocked out a lot of work on a hidden markov model header only
library.  I can start looking into this soon.  Not sure how much thrashing
about I'll end up doing yet, so there might be some extra noise on mailing
lists coming from me.

keryell, 2021-06-18:
> hidden markov model header only library

That sounds like a good idea. You feed your model with the Linux git history to learn about how modernizing the kernel driver API has been done in the past and then it can generate automatically the patch for you... ;-)

chvamshi-xilinx, 2021-07-23:
We tried to fix this , but there are lot of changes required to support 5.11 kernel as drm_driver changed lot of its interfaces . We need to verify so many cases before pushing the changes. This is targeted for next release.

keryell, 2021-07-23:
Good to know someone has looked at it!
I can see now the 2022.1 milestone. This is not the next release, but 2 or 3 releases from now, isn't it?
Is there a branch with the current changes you have?
Thanks.

keryell, 2021-08-02:
Note this is a problem known since 2019...

HFTrader, 2021-08-13:
So just a heads up, XRT is officially broken with respect to Ubuntu 20.04 LTS. 
A recent apt update/upgrade upgraded my kernel to 5.11 and DKMS failed silently.

keryell, 2021-08-14:
Ouch!
@HFTrader thanks for the information.
You are right https://ubuntu.com/about/release-cycle#ubuntu-kernel-release-cycle https://www.phoronix.com/scan.php?page=news_item&px=Ubuntu-20.04.3-LTS-Coming
This means we are in a mess and @sonals needs to put this issue at the top priority.
An interesting example of technical debt in practice, appearing in 2019 and exploding in 2021.

trevor-crowley, 2021-09-19:
work-around:  as read on the Xilinx site, rolling the kernel back to 5.4.0-84-generic works on Ubuntu 20.04.3 LTS

keryell, 2021-09-30:
It looks like there is some progress on this: https://github.com/Xilinx/XRT/pull/5867

keryell, 2021-10-01:
I have just tried today by upgrading our `xsjsycl41` machine: XRT passes now the validation on my U200, Ubuntu 21.04, Linux kernel 5.11!
Let's see next week how it goes with the new Ubuntu 21.10 release... :-)

HFTrader, 2021-10-03:
Has anyone tested it with 20.04.03 LTS?