Xilinx/XRT #5229: Python bindings status

m-kru, 2021-05-10:
Python bindings are available, however they are not documented (except internal source code documentation). Are these bindings stable and usable?

Also going through the Python code I have realized it uses unpythonic  convention to use `xrt` and `xcl` prefixes in names. Why not provide `xrt` and `xcl` modules or packages instead? This is natural way for namespaces in Python.

sonals, 2021-05-20:
The python bindings are maturing and the documentation will be included in the top level XRT doc. 

Would you like to work on creating the xrt module? You can make the changes and provided a Pull Request./

Have you looked into pyxrt pybind11 binding that is coming in 2021.1?

m-kru, 2021-05-21:
@sonals Yes I have looked into it, but I have not used it yet. However, I am sure I will at least try.

I could potentially make the Python code to be more Pythonic. Are you Xilinx authoritative employee able (and willing) to merge?

sonals, 2021-05-27:
Can you provide a pull request with the suggested changes for say two APIs: xrtBOAlloc() and xrtBOFree()? That will help us understand the design.

sonals, 2021-08-10:
Please try the updated XRT. Added support for generating pybind11 documentation. Also going forward the focus will remain on pybind11 as supported python API.

sonals, 2021-11-29:
@m-kru, please try the pybind11 APIs.

closed 2021-11-29 by sonals