Xilinx/XRT #5255: sub-buffer does not work

doonny, 2021-05-17:
Hi, everyone, I am trying to utilize sub-buffer to read a portion of data from a buffer.  I have followed the instructions of 
https://www.xilinx.com/html_docs/xilinx2020_2/vitis_doc/devhostapp.html.
 
The host code is as follows:
```
	// Connect to the desired platform
	platform_id = findPlatform(vendor_name);
	if(platform_id == NULL) {
		printf("ERROR: Unable to find the desired OpenCL platform.\n");
		return EXIT_FAILURE;
	}

	// Query the available OpenCL device
	device.reset(getDevices(platform_id, DEVICE_TYPE, &num_devices));
	printf("Platform: %s\n", getPlatformName(platform_id).c_str());
    printf("Totally %d device(s) are found\n", num_devices);
    device_ptr = 0; // only use one device, select the proper idx
    num_devices = 1; // reset the num of device to 1
	//for(unsigned i = 0; i < num_devices; ++i) {
		printf("  Device %d: %s\n", device_ptr, getDeviceName(device[device_ptr]).c_str());
		displayDeviceInfo(device[device_ptr]);
	//}
	
	// Initialize the data set for each device
	input_a.reset(num_devices);
	output_b.reset(num_devices);
	output_b1.reset(num_devices);

	// Create the context.
	context = clCreateContext(NULL, num_devices, &device[device_ptr], NULL, NULL, &status);
	checkError(status, "Failed to create context");

	// Create Program Objects
	char *kernel_file_name=argv[1];

	// Create the program for all device. All devices execute the same kernel.
	program = createProgramFromFile(context, (const char *) kernel_file_name, &device[device_ptr], num_devices);
	
	// Create per-device objects.
	queue.reset(num_devices);
	kernel.reset(num_devices);
	input_a_buf.reset(num_devices);
	output_b_buf.reset(num_devices);
	output_b1_sbuf.reset(num_devices);

	cl_buffer_region sbuf_info_1={0, DATA_SIZE*sizeof(float)}; // {size_t origin, size_t size}

	for(unsigned i = 0; i < num_devices; ++i) {
		// Command queue
		queue[i] = clCreateCommandQueue(context, device[device_ptr], CL_QUEUE_PROFILING_ENABLE, &status);
		checkError(status, "Failed to create command queue");
		
		// Kernel
		kernel[i] = clCreateKernel(program, kernel_name, &status);
		checkError(status, "Failed to create kernel");
		
		// Input buffers
		input_a_buf[i] = clCreateBuffer(context, CL_MEM_READ_ONLY | CL_MEM_ALLOC_HOST_PTR, 
			DATA_SIZE * sizeof(float), NULL, &status);
		checkError(status, "Failed to create buffer for input A");
		
		// Output buffers
		output_b_buf[i] = clCreateBuffer(context, CL_MEM_WRITE_ONLY | CL_MEM_ALLOC_HOST_PTR,
			DATA_SIZE * sizeof(float), NULL, &status);
		checkError(status, "Failed to create buffer for output B");

		// two sub buffers of the output buffer
		output_b1_sbuf[i] = clCreateSubBuffer(output_b_buf[i], CL_MEM_WRITE_ONLY,
			CL_BUFFER_CREATE_TYPE_REGION, &sbuf_info_1, &status);
		checkError(status, "Failed to create sub buffer for output B");

		// Set kernel arguments.
		unsigned argi = 0;
		unsigned data_num = DATA_SIZE/VEC_SIZE;
		
		status = clSetKernelArg(kernel[i], argi++, sizeof(cl_mem), &input_a_buf[i]);
		checkError(status, "Failed to set argument %d", argi - 1);
	
		status = clSetKernelArg(kernel[i], argi++, sizeof(cl_mem), &output_b_buf[i]);
		checkError(status, "Failed to set argument %d", argi - 1);
		
		status = clSetKernelArg(kernel[i], argi++, sizeof(cl_uint), &data_num);
		checkError(status, "Failed to set argument %d", argi - 1);
        
		// get host ptr
		// XRT will allocate the host buffer automatically, no need to release them
		input_a[i] = (float*)clEnqueueMapBuffer(queue[i], input_a_buf[i], CL_TRUE, CL_MAP_WRITE,
			 0, DATA_SIZE * sizeof(float), 0, NULL, NULL, &status);
		checkError(status, "Failed to map buffer A");
		
		output_b[i] = (float*)clEnqueueMapBuffer(queue[i], output_b_buf[i], CL_TRUE, CL_MAP_READ,
			 0, DATA_SIZE * sizeof(float), 0, NULL, NULL, &status);
		checkError(status, "Failed to map buffer B");

		output_b1[i] = (float*)clEnqueueMapBuffer(queue[i], output_b1_sbuf[i], CL_TRUE, CL_MAP_READ,
			 0, DATA_SIZE * sizeof(float), 0, NULL, NULL, &status);
		checkError(status, "Failed to map sub buffer B");

		// Initialize the data set for each device
		for(unsigned int j = 0; j < DATA_SIZE; j++) {
			input_a[i][j] = rand_float();
		}

		// transfer the data to device
        status = clEnqueueMigrateMemObjects(queue[i], 1, &input_a_buf[i], 0, /* 0 means from host*/
                                            0, NULL, &readevent[i]);
		checkError(status, "Failed to transfer input A");
	}

	// Wait for reads to finish
	clWaitForEvents(num_devices, readevent);
	
	// Execute the kernel

	for(unsigned i = 0; i < num_devices; ++i) {
        
		status = clEnqueueTask(queue[i], kernel[i], 0, NULL, &kernel_event[i]);
        checkError(status, "Failed to launch kernel");
	}
	
	// Wait for all kernel to finish
	clWaitForEvents(num_devices, kernel_event);
	
	// Read back the results from the device to verify the output
	//
	for(unsigned i = 0; i < num_devices; ++i) {

        status = clEnqueueMigrateMemObjects(queue[i], 1, &output_b_buf[i],
                                            CL_MIGRATE_MEM_OBJECT_HOST, 0, NULL, &readevent[i]);
		checkError(status, "Failed to set transfer output B");

        status = clEnqueueMigrateMemObjects(queue[i], 1, &output_b1_sbuf[i],
                                            CL_MIGRATE_MEM_OBJECT_HOST, 0, NULL, &readevent[i]);
		checkError(status, "Failed to set transfer output B");
	}
    
	// Wait for reads to finish
	clWaitForEvents(num_devices, readevent);
```
The kernel code is very simple, it only bypasses the data from input buffer to output buffer

```
typedef ap_uint<VEC_SIZE*32> float_vec;

extern "C" {
void memAccess(
                const float_vec *A_in,
                float_vec *B_out,
				const uint data_num
				)
{
	float_vec tmp;

	for(uint i=0; i<data_num; i++){
		tmp = A_in[i];
		B_out[i] = tmp;
	}

}
}
```

However, the output_b_buf works correctly, but the content of output_b1_sbuf is all zeros.

I am using vitis 2020.1 and tested on both u50 and zc706 boards.

stsoe, 2021-05-19:
@doonny  Thank you for pointing this out.   There is indeed a bug that fails to update the host buffer associated with the sub buffer.

I want to point out though, that since, in your example, the parent buffer is allocated with CL_MEM_ALLOC_HOST_PTR, the sub buffer also will be allocated with this flag (it is inherited per spec).  This means the sub-buffer will not only be offsetting into the parent buffer (device and host ptr) it will also have its own allocated host buffer that must be memcpy'ed with data from the parent host buffer after sync.  This is wasteful.

Typically when sub-buffers have been used with XRT, the user code allocates the parent buffer without any special flags.   This implies that when the buffers are mapped they will be the same virtual address (plus offset for sub buffer).  In your example, the sub buffer migrate (from device->host) would  no longer be needed, or if you only care about the sub buffer content, then the first migrate parent buffer migrate would not be needed.

I will close this issue when a fix for the missing host buffer memcpy has been added.

doonny, 2021-05-19:
@stsoe thx for the use instructions.

PR #5280 (2021-05-19): Issue #5255 : Sub-buffer doesn't work
Add missing memcpy when sub-buffer's host ptr is not shared with
parent buffer.  See comments #5255.

Fixes #5255
commit b4919ee4 (2021-05-19): Issue #5255 : Sub-buffer doesn't work (#5280)

Add missing memcpy when sub-buffer's host ptr is not shared with
parent buffer.  See comments #5255.

Fixes #5255
closed 2021-05-19 by stsoe