Xilinx/XRT #5256: Segmentation fault with `pyopencl`

sgpthomas, 2021-05-17:
`pyopencl` causes a segmentation fault upon import because `libxilinxopencl.so` doesn't define `clSetProgramSpecializationConstant`. Here's how I reproduced this:

```
>>> import pdb
>>> pdb.set_trace()
(pdb) import pyopencl as cl
*** ImportError: /home/sgt43/.local/lib/python3.6/site-packages/pyopencl/_cl.cpython-36m-x86_64-linux-gnu.so: undefined symbol: clSetProgramSpecializationConstant
```

This is my XRT version info:
```
{
  "BUILD_VERSION" : "2.8.743",
  "BUILD_VERSION_DATE" : "Mon, 16 Nov 2020 00:19:11 -0800",
  "BUILD_BRANCH" : "2020.2",
  "VERSION_HASH" : "77d5484b5c4daa691a7f78235053fb036829b1e9",
  "VERSION_HASH_DATE" : "Fri, 13 Nov 2020 07:51:30 -0800"
}
```

sonals, 2021-05-18:
Can you provide more details:

1. What Linux distribution and version are you using?
2. Are you using the OpenCL ICD loader that came with your distribution?
3. Did the PyOpenCL come with the Linux distribution?
4. XRT support OCL 1.2 with some features of 2.0. Normally the ICD loaders do not fail if OCL symbols are missing in the vendor library. clSetProgramSpecializationConstant() was added in 2.2. Is there a way to force PyOpenCL to not look for symbols introduced in 2.2?

sgpthomas, 2021-05-18:
I think it was an issue with some other python environment stuff. I tried in a `virtualenv` wrapper and it worked fine. Sorry for the bother.

closed 2021-05-18 by sgpthomas