Xilinx/XRT #5276: python_binding (SEGFAULT)

JianmingTONG, 2021-05-19:
Hi

I'm using the latest master branch with commit id ```edeb730b9ba2b1d9d36d4bbc99ef454be0264246```.
- UBUNTU 18.04
- cmake 3.20.0
- pybind11 [https://github.com/pybind/pybind11] master branch commit id (f61855b9d8821e2576960109a2a67379a6c2366f)
- python3.7

I'm running into an issue when building the package using the latest version of the XRT library, as shown below:

```
Test project /home/jimmy/work/work_tushar/FPGA_framework/XRT/build/Debug
      Start  1: xbmgmt
 1/10 Test  #1: xbmgmt ............................   Passed    0.00 sec
      Start  2: xbutil
 2/10 Test  #2: xbutil ............................   Passed    0.00 sec
      Start  3: xbutil2
 3/10 Test  #3: xbutil2 ...........................   Passed    0.00 sec
      Start  4: xbmgmt2
 4/10 Test  #4: xbmgmt2 ...........................   Passed    0.00 sec
      Start  5: python_binding
 5/10 Test  #5: python_binding ....................***Exception: SegFault  0.13 sec

      Start  6: [xclbinutil]:signing-xclbin_CER
 6/10 Test  #6: [xclbinutil]:signing-xclbin_CER ...   Passed    0.35 sec
      Start  7: [xclbinutil]:signing-xclbin_DER
 7/10 Test  #7: [xclbinutil]:signing-xclbin_DER ...   Passed    0.34 sec
      Start  8: [xclbinutil]:smartnic-syntax
 8/10 Test  #8: [xclbinutil]:smartnic-syntax ......   Passed    0.48 sec
      Start  9: [xclbinutil]:smartnic-format
 9/10 Test  #9: [xclbinutil]:smartnic-format ......   Passed    0.27 sec
      Start 10: [xclbinutil]:softkernel
10/10 Test #10: [xclbinutil]:softkernel ...........   Passed    0.27 sec

90% tests passed, 1 tests failed out of 10

Total Test time (real) =   1.85 sec

The following tests FAILED:
          5 - python_binding (SEGFAULT)
```

I've tested the pybind11 library works using this repo(https://github.com/pybind/python_example). Might I get any help on how to fix it? 

Thanks in advance!
JT

sonals, 2021-05-20:
Was pybind11 installed using pip3? This has been tested with pybind11 that comes with pip3 and gets installed into /usr/local/lib
Also look inside the test log file generated by cmake to see why the test failed--
cat build/Debug/Testing/Temporary/LastTest.log

JianmingTONG, 2021-05-20:
Thanks for your reply

I install the pybind11 by following the following instruction and use this repo. And I install this repo to satisfy both CMake requirement and python requirement as following:

I'm confused that if I delete the pybind11 in /usr/local/lib then the CMake will complain about the following issue. Might I ask how to satisfy the CMake requirement without having pybind lib files in ``` /usr/local/lib```?

# CMake requirement
```
CMake Error at python/pybind11/CMakeLists.txt:17 (find_package):
  Could not find a package configuration file provided by "pybind11"
  (requested version 2.6.0) with any of the following names:

    pybind11Config.cmake
    pybind11-config.cmake
```

```
git clone https://github.com/pybind/pybind11
cd pybind11
mkdir build 
cd build
make -j6
sudo make install
``` 

# For python requirement:
```
pip3 install pybind11
```

# Here is the log for the python_binding test.

```
5/10 Testing: python_binding
5/10 Test: python_binding
Command: "/usr/bin/python3" "/home/jimmy/work/XRT/src/../tests/python/200_binding/200_main.py"
Directory: /home/jimmy/work/XRT/build/Debug
"python_binding" start time: May 19 22:25 CST
Output:
----------------------------------------------------------
<end of output>
Test time =   0.14 sec
----------------------------------------------------------
Test Failed.
"python_binding" end time: May 19 22:25 CST
"python_binding" time elapsed: 00:00:00
----------------------------------------------------------

6/10 Testing: [xclbinutil]:signing-xclbin_CER
6/10 Test: [xclbinutil]:signing-xclbin_CER
Command: "/bin/bash" "-c" "/home/jimmy/work/XRT/build/Debug/unit-tests/xclbinutil/signing-xclbin_CER/runme.sh"
Directory: /home/jimmy/work/XRT/build/Debug/unit-tests/xclbinutil/signing-xclbin_CER
"[xclbinutil]:signing-xclbin_CER" start time: May 19 22:25 CST
Output:
----------------------------------------------------------
```

Might I also get some tips on what to do to solve the problem?

Thanks in advance!
JT

sonals, 2021-05-20:
For XRT use you do not need to clone pybind11. You can simply install pybind11 using pip3. The problem is unique to your setup. Can you remove the copy of pybind11 that you have installed by cloning and building it from https://github.com/pybind/pybind11?

sonals, 2021-06-01:
@JianmingTONG , let me know if you are good now. I would close the issue.

JianmingTONG, 2021-06-02:
Thanks, the problem is still there.

Because if I only install pybind 11 using pip3. Then it will report

```
CMake Error at python/pybind11/CMakeLists.txt:17 (find_package):
  Could not find a package configuration file provided by "pybind11"
  (requested version 2.6.0) with any of the following names:

    pybind11Config.cmake
    pybind11-config.cmake
```

The only way I know to solve it is to clone the pybind 11 and build locally then followed by installation.

Might I get some suggestions

Thanks again for your kind help and consideration.

sonals, 2021-06-04:
Are you using native cmake? Ubuntu 18.04 cmake version is 3.10 but you are using cmake 3.20.0. This is not a tested configuration.

If you want to use your cmake you may have to tweak the following makefile to help cmake look for the missing files/versions:
`src/python/pybind11/CMakeLists.txt `

sonals, 2021-06-22:
Python integration works on supported platform configurations.

closed 2021-06-22 by sonals

JianmingTONG, 2021-07-03:
Thanks, even though it does not work on my side, it does not affect the correctness of generating xmodel for FPGA. So I just ignore it.