Xilinx/XRT #5288: System Freeze

milo168, 2021-05-21:
Hi, I am trying to use the [clPollStream](https://github.com/Xilinx/XRT/blob/05c9c23f38a111998a5adcb324fdbe681e9d9780/src/include/1_2/CL/cl_ext_xilinx.h#L350) option. During configuration inside the host code, I set the STREAM_OPT_AIO_MAX_EVENT to be 65536 for the stream I want to poll with [clSetStreamOpt](https://github.com/Xilinx/XRT/blob/05c9c23f38a111998a5adcb324fdbe681e9d9780/src/include/1_2/CL/cl_ext_xilinx.h#L366) . I then issue couple tens of thousands (~35000) non-blocking writes to the stream and wait for the tens of thousand events using clPollStream.  After some time, clPollStream never returns (a logical error in my code) and at this point I issue a SIGINT via Crtl+C. The system then freezes and is unresponsive until a hard reboot.

Below are some configuration information.

-  OS: Ubuntu 20.04.02 LTS
- Device: xilinx_u200_qdma_201920_1
-  XRT: 2.9.317 b0230e59e22351fb957dc46a6e68d7560e5f630c
- [dmesg filtered](https://pastebin.com/9zZ1FiDW)
- xrt.ini
```
[Debug]
profile=true
timeline_trace=true
stall_trace=all
data_transfer_trace=coarse

[Emulation]
timeout_scale=ms
debug_mode=batch
```

sonals, 2021-05-28:
Streaming APIs are being deprecated starting 2021.1. Please migrate to Slave Bridge (Host Memory) based design.

milo168, 2021-05-28:
Understand. What are the implications for the QDMA subsystem in the future? Will new APIs be created for it?

sonals, 2021-06-04:
QDMA is supported in MM mode. Streaming interfaces are not supported in XRT/Alveo.

closed 2021-06-24 by milo168