Xilinx/XRT #5401: Kernel address space size hardcoded to 64k

joosthooz, 2021-06-10:
It is currently not possible to use a Vitis RTL kernel with an address space size larger than 64k, because it seems to be hardcoded in xrt:

https://github.com/Xilinx/XRT/blob/6f18dad00c37748db3a7c9b940cce47a1603bdf1/src/runtime_src/core/common/api/xrt_kernel.cpp#L595

The error is `[XRT] ERROR: Cannot read or write outside kernel register space`
My XRT version info:
```
XRT build version: 2.8.743
Build hash: 77d5484b5c4daa691a7f78235053fb036829b1e9
Build date: 2020-11-16 08:21:45
Git branch: 2020.2
```

The address space size is properly listed in the xclbin file:
```
Ports
-----
   Port:          s_axi_control
   Mode:          slave
   Range (bytes): 0x80000
   Data Width:    32 bits
   Port Type:     addressable

   Port:          m00_axi
   Mode:          master
   Range (bytes): 0xFFFFFFFFFFFFFFFF
   Data Width:    256 bits
   Port Type:     addressable
```

stsoe, 2021-06-12:
Investigating.  Normally the address range is part of of the xml meta data in xclbin, however this show 4k by default.

joosthooz, 2021-06-21:
Is there something I can do to help resolving this? I guess a fix would involve more than a simple patch...

stsoe, 2021-06-21:
@joosthooz .  Please confirm what the address range is in the xclbin embedded meta data section.
% xclbinutil --dump-section EMBEDDED_METADATA:raw:<file> --input <xclbin>
Check for the value of `range` attribute in the xml file as shown here:
```
<kernel ...>
  ...
 <port name="S_AXI_CONTROL" mode="slave" range="0x1000" dataWidth="32" portType="addressable" base="0x0"/>
```
The hardcoded part of XRT will be changed to get the range from here.

joosthooz, 2021-06-21:
Hi,
This is the requested lines from the dump of my xclbin:
```
<kernel name="hbm_tta_vec2add" language="ip_c" vlnv="tud_tut:kernel:hbm_tta_vec2add:1.0" preferredWorkGroupSizeMultiple="0" workGroupSize="1" debug="true" interrupt="false" hwControlProtocol="ap_ctrl_hs">
          <port name="s_axi_control" mode="slave" range="0x80000" dataWidth="32" portType="addressable" base="0x0"/>
```

joosthooz, 2021-07-05:
Hi,
Thanks for merging a fix. I wanted to try it so I built xrt (master) from source and linked my app to the new library, but then the whole server froze when the app tried to load the bitstream... Unfortunately I don't have any logs or dumps that survived the crash. What could have caused this? Should I wait for a new release and then install the complete XRT instead?

stsoe, 2021-07-08:
Hi @joosthooz .  Without access to xclbin and host code it is difficult to say what is causing the crash you are observing.
Is it possible to get access to xclbin and host code?

joosthooz, 2021-07-12:
Hi,
Please find the xclbin file here: https://bitbucket.org/joosthooz/tta-simd-streamingio/src/crashing_bitstream/xclbin_hw_build/hbm_tta_vec2add.xclbin
and the host code here:
https://bitbucket.org/joosthooz/tta-simd-streamingio/src/crashing_bitstream/vivado_srcs/hbm_tta_vec2add/exports/src/host_nativeAPI.cpp

The 64k limit is still in effect when targeting `hw_emu` (see https://github.com/Xilinx/XRT/tree/master/src/runtime_src/core/pcie/emulation/hw_em/generic_pcie_hal2/shim.h#L387)

stsoe, 2021-07-12:
@joosthooz Thanks for providing access to xclbin and host code.

I looked at host_nativeAPI.cpp and first problem appears to be that it allocates buffers in memory bank "4" which is not connected per xclbin.     It's btw preferable to use the C++ APIs  which manages lifetime and throws on error; the C-API have to  be checked for errors.

Please additionally provide how to run the host code, e.g. what and where is the tta_program?

joosthooz, 2021-07-13:
Hi,
Thank you for taking the time to look into this, I really appreciate it!
I started out with the example host code that was generated by Vivado (the RTL Kernel wizard). But because I want to manually control my kernel, I've converted it to using the native API. I will look into using the C++ API, thanks.

The strange memory bank indexes were in the generated example code. It is a mystery to me where they came from. That's why I left them the way they were.

Preferably I'd want to use an API to query the xclbin file what memory bank each port is connected to. It seems when using the C++ API I can use xrt::kernel::group_id() for this (as documented here: https://xilinx.github.io/XRT/master/html/xrt_native_apis.html#getting-bank-group-index-of-the-kernel-argument), but it is not clear to me if this will work the same way for user-managed kernels (because here: https://xilinx.github.io/XRT/master/html/xrt_native_apis.html#allocating-buffers-for-the-ip-inputs-outputs
it instructs to use `xbutil examine --report memory`).

The tta_program is a binary that I want to load into write into an instruction memory that is mapped into the kernel's control address space. As far as I could tell, the crash seemed to occur already during loading the bitstream, before the kernel was being accessed.
To execute the host code: `./hbm_tta_vec2add_host_example.exe ../../../xclbin_build/hbm_tta_vec2add.xclbin hbm_tta_vec2add 1 ../../../vec2add.img`

stsoe, 2021-07-13:
Hi @joosthooz 

No problem.  Your case is good to verify that the expanded register read/write works as expected.   Besides the hw emulation limitation you pointed out, everything is good.  The problem appears to be in the host code at this point.

For the memory connectivity; you are trying to allocate buffers for the 3 kernel global arguments, kernel arg no 4, 5, and 6.  You must allocate the buffers in the banks that are connected to the corresponding kernel arguments.  You do this by using xrt::kernel::group_id(argno) or xrtKernelArgGroupId(kernel_handle, argno) with the C-APIs.   Once you have the group id (which really is just an identifier for the connected bank), you proceed using this computed value rather than the hard code one.   

```
    auto bank4 = xrtKernelArgGroupId(platform.kernel, 4); // argno 4
    buf00.handle = xrtBOAlloc(platform.device, size, XRT_BO_FLAGS_NONE, bank4);
    ...
    auto bank5 = xrtKernelArgGroupId(platform.kernel, 5); // argno 5
    buf01.handle = xrtBOAlloc(platform.device, size, XRT_BO_FLAGS_NONE, bank5);
    ...
    auto bank6 = xrtKernelArgGroupId(platform.kernel, 6); // argno 6
    buf02.handle = xrtBOAlloc(platform.device, size, XRT_BO_FLAGS_NONE, bank6);
```
This will get you over the first hump.   The second problem is dereferencing garbage in this call:

`      rc = writeMMIO(writeAddress, *((uint32_t*)mapped_tta_prog[i]));`

Please let me know if I can help with any specific XRT questions you run into.  I appreciate you pointing out the earlier 64k limit and the hw emulation limit also.   I will file a change request to get hw emulation fixed.

I will close this issue once hw emulation is fixed unless you identify other related problems.

Thank you.

stsoe, 2021-07-13:
Change request CR-1105172 created for HW emulation support of dynamic address range.

stsoe, 2021-08-10:
@venkatp-xilinx Please mark this issue fixed when CR-1105172 is fixed.  Thanks!

stsoe, 2021-09-09:
HW emulation support in https://github.com/Xilinx/XRT/pull/5710.
Closing this issue as it should be fixed now.

stsoe, 2021-09-10:
@joosthooz, the dump you show looks correct.  Could you by any chance either provide the xclbin or debug and catch the throw with a stack trace?

joosthooz, 2021-09-11:
Hi,
I pushed my xclbin here:
https://bitbucket.org/joosthooz/tta-simd-streamingio/src/range_not_found_xclbin/xclbin_build/hbm_tta_vec2add.xclbin
I don't know if it has any influence, but it was created with the new XRT sourced. It is a hw_emu bitstream.
The way I'm running it is from the `vivado_srcs/hbm_tta_vec2add/exports` dir, build the runtime with `make` and then
```
export XCL_EMULATION_MODE=hw_emu
./hbm_tta_vec2add_host_example.exe ../../../xclbin_build/hbm_tta_vec2add.xclbin hbm_tta_vec2add 1 ../../../vec2add.img
```

joosthooz, 2021-09-11:
I'm still using the C API so no exception to catch...

stsoe, 2021-09-14:
The emulation fix is not correct.  @venkatp-xilinx see the CR

stsoe, 2021-09-19:
@joosthooz .  Should be fixed now with https://github.com/Xilinx/XRT/pull/5808

closed 2021-09-19 by stsoe