Xilinx/XRT #5517: XRT_docs

durgabhavaniv, 2021-07-22:
I am not able to generate XRT_docs.

command used : 
./build.sh -docs -ertfw /opt/xilinx/xrt/share/fw/sched.bin
Facing issue:
make xrt_docs
make: *** No rule to make target 'xrt_docs'.  Stop.

durgabhavaniv, 2021-07-22:
Hi able to fix this but facing issue as follows:
[image]

uday610, 2021-07-29:
@durgabhavaniv , document generation may not outside Xilinx. We don't support doc generation flow outside Xilinx. 

XRT team generates document and publish here: https://xilinx.github.io/XRT/master/html/index.html , so you can look into these HTML files if you require any information.

uday610, 2021-08-02:
closed as per previous comment

closed 2021-08-02 by uday610