Xilinx/XRT #5539: XRT 2021.1 build ERROR

HSqure, 2021-07-28:
### Operation Describe

According  [XRT Xilinx Docs](https://xilinx.github.io/XRT/master/html/build.html#building-the-xrt-installation-package) complete the *Installing Building Dependencies* checking.

### Environment

`Ubuntu 18.04`
`Cmake 3.10.2`

### ERROR Log

```
alinx@server:~/huahua/workspace/Vitis-AI-master-v1.4/setup/XRT/build$ ./build.sh 
cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_BUILD_TYPE=Debug ../../src
-- Host system processor is x86_64
-- Target system processor is x86_64
-- Looking for DRM - found at /usr 2.4.101
-- Looking for OPENCL - found at /usr 2.1 /usr/include
-- Looking for GIT - found at /usr/bin/git
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
-- XRT EA eula files  /home/alinx/huahua/workspace/Vitis-AI-master-v1.4/setup/XRT/src/../LICENSE
-- Platform/Linux (Ubuntu) (Kernel 5.4.0-72-generic)
-- Compiler: /usr/bin/c++ /usr/bin/cc
CMake Error at runtime_src/tools/xclbinutil/CMakeLists.txt:22 (find_package):
  By not providing "FindRapidJSON.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "RapidJSON", but CMake did not find one.

  Could not find a package configuration file provided by "RapidJSON" with
  any of the following names:

    RapidJSONConfig.cmake
    rapidjson-config.cmake

  Add the installation prefix of "RapidJSON" to CMAKE_PREFIX_PATH or set
  "RapidJSON_DIR" to a directory containing one of the above files.  If
  "RapidJSON" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/alinx/huahua/workspace/Vitis-AI-master-v1.4/setup/XRT/build/Debug/CMakeFiles/CMakeOutput.log".

real	0m0.137s
user	0m0.097s
sys	0m0.040s

```
What should do to solve this problem to install the xrt? NEED HELP!

Thanks!

uday610, 2021-07-29:
@HSqure , have you run xrtdeps.sh (require sudo) as mentioned in the link? 

If you already executed xrtdeps.sh by sudo, can you logout from machine and login again and try to build?

HSqure, 2021-07-29:
> @HSqure , have you run xrtdeps.sh (require sudo) as mentioned in the link?
> 
> If you already executed xrtdeps.sh by sudo, can you logout from machine and login again and try to build?

Thanks for replying! 

Yeah, I executed `xrtdeps.sh` with sudo and here's log below:

```
alinx@server:~/huahua/workspace/Vitis-AI-master-v1.4/setup/XRT$ sudo src/runtime_src/tools/scripts/xrtdeps.sh
Preparing ubuntu ...
Installing packages...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libboost-dev is already the newest version (1.65.1.0ubuntu1).
libboost-filesystem-dev is already the newest version (1.65.1.0ubuntu1).
libboost-program-options-dev is already the newest version (1.65.1.0ubuntu1).
libjpeg-dev is already the newest version (8c-2ubuntu8).
libprotoc-dev is already the newest version (3.0.0-9.1ubuntu1).
libyaml-dev is already the newest version (0.1.7-2ubuntu3).
lsb-release is already the newest version (9.20170808ubuntu1).
make is already the newest version (4.1-9.1ubuntu1).
ocl-icd-dev is already the newest version (2.2.11-1ubuntu1).
ocl-icd-libopencl1 is already the newest version (2.2.11-1ubuntu1).
ocl-icd-opencl-dev is already the newest version (2.2.11-1ubuntu1).
pkg-config is already the newest version (0.29.1-0ubuntu2).
python3-sphinx is already the newest version (1.6.7-1ubuntu1).
sphinx-common is already the newest version (1.6.7-1ubuntu1).
strace is already the newest version (4.21-1ubuntu1).
cppcheck is already the newest version (1.82-1).
gnuplot is already the newest version (5.2.2+dfsg1-2ubuntu1).
graphviz is already the newest version (2.40.1-2).
libgtest-dev is already the newest version (1.8.0-6).
lm-sensors is already the newest version (1:3.4.0-4).
opencl-headers is already the newest version (2.2~2018.02.21-gb5c3680-1).
protobuf-compiler is already the newest version (3.0.0-9.1ubuntu1).
python3-breathe is already the newest version (4.7.3-1).
python3-sphinx-rtd-theme is already the newest version (0.2.4-1).
libjson-glib-dev is already the newest version (1.4.2-3ubuntu0.18.04.1).
python3 is already the newest version (3.6.7-1~18.04).
curl is already the newest version (7.58.0-2ubuntu3.13).
dmidecode is already the newest version (3.1-1ubuntu0.1).
libdrm-dev is already the newest version (2.4.101-2~18.04.1).
libssl-dev is already the newest version (1.1.1-1ubuntu2.1~18.04.9).
unzip is already the newest version (6.0-21ubuntu1.1).
linux-libc-dev is already the newest version (4.15.0-142.146).
linux-libc-dev set to manually installed.
libtiff5-dev is already the newest version (4.0.9-5ubuntu0.4).
uuid-dev is already the newest version (2.31.1-0.4ubuntu3.7).
gcc is already the newest version (4:7.4.0-1ubuntu2.3).
perl is already the newest version (5.26.1-6ubuntu0.5).
libncurses5-dev is already the newest version (6.1-1ubuntu1.18.04).
linux-headers-5.4.0-72-generic is already the newest version (5.4.0-72.80~18.04.1).
linux-headers-5.4.0-72-generic set to manually installed.
dkms is already the newest version (2.3-3ubuntu9.7).
libopencv-core-dev is already the newest version (3.2.0+dfsg-4ubuntu0.1).
python3-pip is already the newest version (9.0.1-2.3~ubuntu1.18.04.4).
pciutils is already the newest version (1:3.5.2-1ubuntu1.1).
g++ is already the newest version (4:7.4.0-1ubuntu2.3).
libsystemd-dev is already the newest version (237-3ubuntu10.44).
libpng-dev is already the newest version (1.6.34-1ubuntu0.18.04.2).
cmake is already the newest version (3.10.2-1ubuntu2.18.04.1).
libudev-dev is already the newest version (237-3ubuntu10.44).
gdb is already the newest version (8.1.1-0ubuntu1).
git is already the newest version (1:2.17.1-1ubuntu0.8).
libpython3-dev is already the newest version (3.6.7-1~18.04).
libpython3-dev set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libcurl4-openssl-dev : Depends: libcurl4 (= 7.58.0-2ubuntu3) but 7.58.0-2ubuntu3.13 is to be installed
 libdw-dev : Depends: libdw1 (= 0.170-0.4) but 0.170-0.4ubuntu0.1 is to be installed
 libelf-dev : Depends: libelf1 (= 0.170-0.4) but 0.170-0.4ubuntu0.1 is to be installed
E: Unable to correct problems, you have held broken packages.
Requirement already up-to-date: pybind11 in /home/alinx/.local/lib/python3.6/site-packages
```
I'll try to re-logging.

HSqure, 2021-07-29:
Hi, @uday610 .

The problem has been solved! According to:
```
libcurl4-openssl-dev : Depends: libcurl4 (= 7.58.0-2ubuntu3) but 7.58.0-2ubuntu3.13 is to be installed
 libdw-dev : Depends: libdw1 (= 0.170-0.4) but 0.170-0.4ubuntu0.1 is to be installed
 libelf-dev : Depends: libelf1 (= 0.170-0.4) but 0.170-0.4ubuntu0.1 is to be installed
```
I reinstalled these dependencies one by one for the specified version and it works.
Thanks again.

closed 2021-07-29 by HSqure