Xilinx/XRT #5560: Add support for KV260 edge platform (and recipe/instructions for adding other edge platforms)

vmayoral, 2021-07-31:
Hello all,

I was hoping to test some XRT-powered acceleration kernels with KV260 when I bumped into the following errors:

```bash
XRT build version: 2.8.0
Build hash: b94857f15ba8c8251df446e8c51af7e0a7c9e061
Build date: 2021-06-11 07:18:33
Git branch: 2020.2
PID: 1694
UID: 0
[Sat Jul 31 16:33:09 2021 GMT]
HOST: xilinx-k26-starterkit-2020_2.2
EXE: /ros2_ws/lib/faster_vadd_publisher/faster_vadd
[XRT] ERROR: No devices found
[XRT] ERROR: No devices found
[XRT] ERROR: No devices found
ERROR: Failed to find Xilinx platform
```

When introspecting the system with `xbutil`, I realized things were certainly wrong:

<details><summary>xbutil introspection of KV260</summary>

```bash
xilinx-k26-starterkit-2020_2:/ros2_ws/lib/faster_vadd_publisher# xbutil list
INFO: Found total 1 card(s), 1 are usable
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
System Configuration
OS name:	Linux
Release:	5.4.0-xilinx-v2020.2
Version:	#1 SMP Thu Jun 10 22:03:38 UTC 2021
Machine:	aarch64
Glibc:		2.30
Distribution:	N/A
Now:		Sat Jul 31 16:40:06 2021 GMT
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
XRT Information
Version:	2.8.0
Git Hash:	b94857f15ba8c8251df446e8c51af7e0a7c9e061
Git Branch:	2020.2
Build Date:	2021-06-11 07:18:33
ZOCL:		2.8.0,b94857f15ba8c8251df446e8c51af7e0a7c9e061
Failed to open device[0]
ERROR: Card index 0 is out of range

xilinx-k26-starterkit-2020_2:/ros2_ws/lib/faster_vadd_publisher# xbutil status
INFO: Found total 1 card(s), 1 are usable
Failed to open device[0]
ERROR: Card index 0 is out of range

xilinx-k26-starterkit-2020_2:/ros2_ws/lib/faster_vadd_publisher# xbutil validate
Unsupported API

xbutil query
INFO: Found total 1 card(s), 1 are usable
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
System Configuration
OS name:	Linux
Release:	5.4.0-xilinx-v2020.2
Version:	#1 SMP Thu Jun 10 22:03:38 UTC 2021
Machine:	aarch64
Glibc:		2.30
Distribution:	N/A
Now:		Sat Jul 31 16:43:10 2021 GMT
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
XRT Information
Version:	2.8.0
Git Hash:	b94857f15ba8c8251df446e8c51af7e0a7c9e061
Git Branch:	2020.2
Build Date:	2021-06-11 07:18:33
ZOCL:		2.8.0,b94857f15ba8c8251df446e8c51af7e0a7c9e061
Failed to open device[0]
ERROR: Card index 0 is out of range
```

</details>

The official KV260 platforms are available at https://github.com/Xilinx/kv260-vitis but none of these are listed at https://github.com/Xilinx/Vitis_Embedded_Platform_Source. Moreover, according to https://github.com/Xilinx/XRT/blob/master/CHANGELOG.rst:

> XRT now has production support for edge platforms. The following non DFX platforms edge platforms are supported: zcu102_base, zcu104_base, zc702, zc706. In addition zcu102_base_dfx platform has DFX support.

Which led me to realize that KV260 is probaly not supported by XRT at this point. 

How can one go ahead and add support for a new edge platform? Particularly, any of the [official KV260 platforms](https://github.com/Xilinx/kv260-vitis/tree/release-2020.2.2_k26/platforms)?

vmayoral, 2021-08-02:
Kindly pinging maintainers @stsoe, @sonals, @chienwei-lan, @mamin506, any chance you could provide feedback on this? 

I'm being told that XRT has been used to produce Kria K26 app (e.g. [smartcam](https://github.com/Xilinx/smartcam)) as a dependency of [Vitis-AI](https://github.com/Xilinx/Vitis-AI). I scanned the source code of smartcam and didn't see a clear XRT connection. The output of `xbutil` also makes me think that XRT needs a few patches to support KV260 boards.

I debugged things in a [simple example](https://gist.github.com/vmayoral/884c850b037871c67bc3f5966330fe36#file-faster_vadd-cpp-L65) and traced things down to `cl::Platform::get(...)` calls. This piece of code fails with XRT recipes shipped in PetaLinux 2020.2.2 BSP. Can anyone share their view? Alternatively, maybe someone more familiar with the code can give me a few pointers on where does this logic lives inside XRT?

Happy to do further debugging but I need support on how to enable KV260.

chvamshi-xilinx, 2021-08-04:
Can you run xbutil2 examine -r host. Looks like zocl module is not loaded.

vmayoral, 2021-08-04:
@chvamshi-xilinx was extremely helpful debugging this and providing a solution so kudos for him! With his help, we found the non-trivial (at least to me) issue. In a nutshell, though the file system had all the necessary components, the device tree didn't contain the necessary entry for ZOCL. This led to an incomplete embedded setup, with some missing directories under `/dev`. Fixing the device tree does it.

**NOTE**: This is caused by issues within the `2020.2.2` PetaLinux BSP release. `2021.1` should address these issues. Leaving here details of our actions and fix for the device tree:

-----

Get the sources from the device tree blob:
```bash
dtc -I dtb -O dts -o system.dts system.dtb
```
Edit `system.dts` and add the following:
```
                zyxclmm_drm {
                        compatible = "xlnx,zocl";
                        status = "okay";
                };
```
Build again the device tree into its blob:
```bash
dtc -I dts -O dtb -o system.dtb system.dts
```

closed 2021-08-04 by vmayoral