Xilinx/XRT #5650: Empty profiling traces when using pybind11 XRT API

strocode, 2021-08-19:
Hi,

I have an XCLBIN build for U280 I want to profile. 

If I drive it with C++ host code via the OpenCL API I get profiling information I can view with vitis_analyzer. I'm most interested in kernel enqueues:
[image]

If I drive it with python via pybind11 XRT API it produces a new device_trace_0.csv, device_trace_1.csv and xclbin.run_summary, but there are no kernel enqueues:

[image]

Is profiling of kernel enqueues supported for pybind11 host applications? The [Vitis 2020.2 profiling documenatation](https://www.xilinx.com/html_docs/xilinx2020_2/vitis_doc/profilingapplication.html) seems to suggest profiling "XRT native API call sequences" is supported. Do I need to do something to turn it on?

My xrt.ini

```
[Debug]
profile=true
timeline_trace=true
data_transfer_trace=fine
stall_trace=all
opencl_summary=true
opencl_trace=true
#trace_buffer_size=1G
[Emulation]
#debug_mode=batch
debug_mode=gui
```

XRT version:
```
XRT build version: 2.11.634
Build hash: 5ad5998d67080f00bca5bf15b3838cf35e0a7b26
Build date: 2021-06-08 22:10:49
Git branch: 2021.1
```

uday610, 2021-08-24:
Hello @strocode , I see you are using the latest 21.1 XRT, which is good. Can you please use the following switch in the xrt.ini file? Please remove all other switches under [Debug] and first try with only the switch below and see if you see something. I believe this will generate a new file named native_trace.csv

[Debug]
native_xrt_trace=true

uday610, 2021-09-15:
@strocode , have you check this? I am closing this as I did not receive any response. Please feel free to file another issue if you still have problem

closed 2021-09-15 by uday610