Xilinx/XRT #5785: Problem with XRT shell, please check it, thanks

RivenChen, 2021-09-14:
hi, I use default  image in "XRT/src/runtime_src/core/pcie/tools/cloud-daemon/aws/aws_dev.h" (DEFAULT_GLOBAL_AFI), and I can use it to do smiple xdma read&write.

But when I use the others image(through aws tos download), the  smiple xdma read&write can't work.
except shell ,does role ip should do something to support xdma read&write ? If it does,  I want to know the role-ip implementation about this "DEFAULT_GLOBAL_AFI", If it doesn't, Is the difference of shell between images leads to this problem.

Thanks.

RivenChen, 2021-09-14:
@xuhz @maxzhen @kennylau-arista

RivenChen, 2021-09-14:
@xuhz @maxzhen @kennylau-arista 
please check this problem, thanks very much.

xuhz, 2021-09-14:
first of all, I assume you run in a aws f1 instance, instead of pegasus.
secondly, DEFAULT_GLOBAL_AFI is loaded to change the device id of the FPGA, but for your purpose, you can just think of it as loading a helloworld kernel to FPGA through xrt. In this case, xrt has mem topology info of the card, then you can run xdma read&write (xbutil mem read/write)
thirdly, if you load an image with aws tool (fgpa-*) instead of xrt, then xrt doesn't have mem topology info, you can't run xdma test

RivenChen, 2021-09-14:
Yes, I run in a aws F1 instance.
 I use the xdma driver directly not xrt api to test dma read&write, and write data to FPGA ddr. I wondle why I can use xdma driver with xrt image...
Does "DEFAULT_GLOBAL_AFI" helloworld kernel have code? I want to have a check on it  because I guess if my CL(custom logic) lacks some interface to shell. Does the CL still needs to be implemented to get access to the shell ddr?

@xuhz  thanks a lot.

xuhz, 2021-09-14:
xrt (xocl driver) includes the xdma driver. One can use either xrt or xdma driver directly to access xdma IP. Either way, when the dma engine is programmed, we need to tell the engine the physical addr of the ddr. 
If xrt (xocl) is being used, once a design is loaded through xrt, the memory topology (in xclbin metadata) is known to xrt, so we can do dma test. 
Regarding to DEFAULT_GLOBAL_AFI, i don't know whether there is code. 
I guess since the shell is dma shell, the CL only needs the connection to the ddr so that CL can access ddr, but the dma itself between shell ddr and host ddr, the CL doesn't need to implement

RivenChen, 2021-09-14:
thanks  @xuhz   I understand your answer. 
I currently have a requirement to use the xdma driver to run our business. Above the xdma are the user-mode apis that we encapsulate. These apis implement functions similar to the bo cu in xrt.  Can I directly call the read/write of xdma to work with the image compiled by Vitis xrt?

sonals, 2021-09-16:
@RivenChen XRT and Vitis are compatible with each other. Using XDMA driver directly with Vitis compiled design is not supported. Can your host application move to XRT API?

xuhz, 2021-09-17:
@sonals 
i helped him out. He can use xdma driver directly moving data now

RivenChen, 2021-09-17:
@sonals XRT use xdma driver as well. Xdma read/write api can also work when user  gets the memory topology of the fpga ddr, and move data to ddr. But in XRT, memory topology  is analyzed through programing image.
@xuhz help me with this problem, thanks a lot.

closed 2021-09-22 by uday610