Xilinx/XRT #5801: Improve the commit message quality

keryell, 2021-09-16:
Often when I look at commits in this project it is not obvious to figure out what is going on.
For example https://github.com/Xilinx/XRT/pull/5789
It would be nice on this project to have useful commit messages that explains what is going on, why it is done, etc.

Some suggestions I found quickly:
- https://www.freecodecamp.org/news/writing-good-commit-messages-a-practical-guide/
- https://joshuatauberer.medium.com/write-joyous-git-commit-messages-2f98891114c4
- https://dev.to/helderburato/patterns-for-writing-better-git-commit-messages-4ba0

@uday610 @stsoe any idea where it could be stated in the developer documentation and somewhat enforced across the project to have top-notch code quality?

Besides the commit messages, it would be nice to have also useful comments in the code instead of having write-only source code. There are some good suggestions in talks from Kate Gregory like https://www.youtube.com/watch?v=MqugiGzricQ with the idea of coding as an empathetic and inclusive way of communicating.

stsoe, 2021-09-16:
Thank you @keryell for the pointers.  I will take an AI to create a summary on our internal documentation and share with our developers in one of our bi-weekly meetings.  Wholeheartedly agree with you on both commit messages and code commenting.

uday610, 2021-09-16:
@keryell , thanks for reminding these important practices again :)