Xilinx/XRT #5833: Configure "force_program_xclbin" in ini to reset the kernel, but failed

xym341, 2021-09-23:
1,ver
commit 859b95f89315cdd220734ef4bb8240e62e97ea77 (HEAD -> master, origin/master, origin/HEAD)
Date:   Mon Sep 13 15:59:35 2021 -0700
2,demand
I wanted to forcibly reset the kernel, but I didn't see the effect。The user layer is configured as follows：
xrt::ini::set("Runtime.force_program_xclbin", true);
3,debug
I tracked the code and saw the judgment of flag in "src/runtime_src/core/pcie/driver/linux/xocl/userpf/**xocl_ioctl.c**".
_xocl_read_axlf_helper(struct xocl_drm *drm_p, struct drm_xocl_axlf *axlf_ptr){
...
 if (xclbin_downloaded(xdev, &bin_obj.m_header.uuid)) {
                if ((axlf_ptr->flags & XOCL_AXLF_FORCE_PROGRAM)) {
                        // We come here if user sets force_xclbin_program
                        // option "true" in xrt.ini under [Runtime] section
                        DRM_WARN("xnjd %s Force xclbin download", __func__);
                } else {
                        goto done;
                }
        }
}_
However, after the UUID matches in "src/runtime_src/core/pcie/driver/linux/xocl/subdev/icap.c", **the kernel will not be reset.**
___icap_peer_xclbin_download(struct icap *icap, struct axlf *xclbin){
...
        peer_uuid = (xuid_t *)icap_get_data_nolock(icap->icap_pdev, PEER_UUID);
        if (uuid_equal(peer_uuid, &xclbin->m_header.uuid)) {
                ICAP_INFO(icap, "xclbin already on peer, skip downloading");
                return 0;
        }
}_
**It does not have the effect of forced reset**
4,test
4.1. I added the function "icap_set_forcedown_flag" in "icap.c";
4.2. Call "icap_set_forcedown_flag" in "xocl_ioctl.c" to set the flag;
4.3. Force the uuid to be cleared when the flag is detected in "_icap_peer_xclbin_download()" in "icap.c".
**Achieved the purpose of resetting kenel, but I am not sure if this is done. At the same time, I also want to know if Xinlinx has any specific instructions on this piece.**

[image]

rajkumar-xilinx, 2021-09-28:
Fixed this issue using https://github.com/Xilinx/XRT/pull/5858 .  It is merged into XRT master branch.
Please sync to latest XRT and verify. Thank you.

xym341, 2021-09-30:
> Fixed this issue using #5858 . It is merged into XRT master branch. Please sync to latest XRT, verify again. and let us know. Thank you.

Verify again. It's solved. Thank you.

closed 2021-09-30 by xym341