Xilinx/XRT #5856: The question about the IO path from storages to devices

pyhundan, 2021-09-27:
Hi,
    I had read some example codes from  **Vitis_Accel_Examples**. I read one of [codes](https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/host/p2p_bandwidth/src/host.cpp)
   I learned that the interface used to transfer data to the device should be `enqueueMigrateMemObjects` and I read the source code in this repository. Maybe I found that implementation of the `enqueueMigrateMemObjects` in `XRT/src/runtime_src/xocl/api/clenqueueMigrateMemObjects.cpp`. 
   As I continued to read the source code, I found some difficulties.  Whether data migration takes place in this line of [code](https://github.com/Xilinx/XRT/blob/master/src/runtime_src/xocl/api/clEnqueueMigrateMemObjects.cpp#L97)
  If it is true,  and I found that actually use [read_cb](https://github.com/Xilinx/XRT/blob/master/src/runtime_src/xocl/api/plugin/xdp/profile_trace.cpp#L108), but the `dlsym(void* handle, const char* symbol)` seems return `::GetProcAddress(HMODULE(handle),symbol);`. Because of this reason, I couldn't go on my study,
  If it is false, could you point the real implement to me ?
Thanks,
Hundan

stsoe, 2021-09-29:
@pyhundan, correct clEnqueueMigrateMemObjects transfers data to/from device.
The code that does the migration is *not* what you highlighted, but rather one line above at line 96.  When event dependencies are satisfied, the enqueue action is called. 

The enqueue action for migration is xocl::enqueue::action_migrate_memobjects defined [here](https://github.com/Xilinx/XRT/blob/512c814d916029687b930bcb929c59f28621dcd0/src/runtime_src/xocl/api/enqueue.cpp#L451).

pyhundan, 2021-10-01:
Hi, @stsoe 
Thank you for your answer.  Your answer has solved my doubts well, but I still have the following questions to ask you.

1、I've found a function implementation for data migration, but I don't understand [sync()](https://github.com/Xilinx/XRT/blob/512c814d91/src/runtime_src/xocl/core/device.cpp#L741), and I can't find an implementation for this function, which jumps to strange locations.

2、 Another question is about the meaning of some functions, for example [xocl::profile::action_migrate](https://github.com/Xilinx/XRT/blob/master/src/runtime_src/xocl/api/clEnqueueMigrateMemObjects.cpp#L97).
In fact, the function [read_cb ](https://github.com/Xilinx/XRT/blob/master/src/runtime_src/xocl/api/plugin/xdp/profile_trace.cpp#L108)will be called in the end, but this function actually reads the contents of the DLL. I want to ask what the contents of this DLL are.

Thanks,
Hundan

stsoe, 2021-10-01:
@pyhundan The level of detail you asking for are best explored using gdb.  You will find that the sync() call ultimately ends up in a [low-level shim library](https://github.com/Xilinx/XRT/blob/709e4e35f8b965af032c40b408023716e5f4d712/src/runtime_src/core/pcie/linux/shim.cpp#L847)  that makes a call into the kernel driver to do the DMA.

As for 2), this is profiling code that annotates time line for the sync operation.  Again, please explore with gdb if you are interested in this level of detail.

pyhundan, 2021-10-11:
Hi @stsoe ,

I have run the Vitis_Accel_Example such as this [code](https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/hello_world/src/host.cpp)

However, no matter the Vitis compiler is used to debug the code, or GDB is used to debug the code from the command line, but the final error will be reported.

Error notification in VItis is `Can't find a source file at "/opt/xrt/src/runtime_src/xdp/appdebug/appdebug.cpp"`

And Error notification at the end of GDB is 

```
gnu_cxx::new_allocator<cl::Memory>::deallocate (this=<optimized out>, __p=<optimized out>) at /usr/include/c++/5/ext/new_allocator.h:110
110	      { ::operator delete(__p); }
(gdb) 
0x00007ffff77b3f10 in operator delete(void*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) 
Single stepping until exit from function _ZdlPv,
which has no line number information.

__GI___libc_free (mem=0x63c0d0) at malloc.c:2941
2941	malloc.c: No such file or directory.
```

How do I debug to find the correct function call flow?

Thank, hundan

stsoe, 2022-04-13:
Closing, no additional feedback.

closed 2022-04-13 by stsoe