Xilinx/XRT #5974: Error: Failed to find Xilinx platform

Hecmay, 2021-11-07:
Hi,

I am trying to run sw emulation using Vitis on a machine that does not have FPGA devices attached. 

Supposedly you would not need any real FPGA hardware to run emulation. However, I am getting this error from XRT when running this function. XRT complains that `Error: Failed to find Xilinx platform`

```cpp
auto devices = xcl::get_xil_devices();
```

If I switch to another machine with FPGAs installed, then it worked fine. How should I solve the error?

keryell, 2021-11-10:
Just curious about the context. Have you installed XRT with a shell for a specific board?
I use this on my laptop to give demos with the emulator for a U200 without any real board.
Actually not with this `xcl::` host API but the OpenCL one, but it should work the same.

Hecmay, 2021-11-11:
@keryell thanks for your reply.

I was using an XDMA shell for the U280 board. I got the example from Vitis Accel Example repo. So are you using the standard OpenCL runtime APIs instead of the `xcl` APIs by XRT? 

Can you please provide some reference code if possible? appreciate it

keryell, 2021-11-12:
I do not have a simple code sample because I use the board through a framework which happens to use the OpenCL API to control the FPGA.

heeran-xilinx, 2021-12-02:
Hi @Hecmay ,
Can you share which particular example of Vitis Accel Examples you are trying? 

You can find list of Native XRT based example Here:
https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/host_xrt

These examples works for emulation flow without FPGA card. Did you try the Makefile flow provided there? I hope emconfig was also successfully running before emulation run. 

-Heera

Hecmay, 2021-12-02:
@heeran-xilinx thanks for you reply. I tried again with XRT native APIs. The 

```shell
(tests) [user@host-name hello_world_xrt]$ XCL_EMULATION_MODE=sw_emu ./hello_world_xrt -x build_dir.sw_emu.xilinx_u280_xdma_201920_3/vadd.xclbin
Open the device0
Load the xclbin build_dir.sw_emu.xilinx_u280_xdma_201920_3/vadd.xclbin
terminate called after throwing an instance of 'std::runtime_error'
  what():  No ip layout available to construct kernel, make sure xclbin is loaded
```

vishnuchebrolu, 2021-12-08:
Hi @Hecmay ,
I see the design passing locally.
Logs : 
XCL_EMULATION_MODE=sw_emu ./hello_world_xrt -x ./build_dir.sw_emu.xilinx_u280_xdma_201920_3/vadd.xclbin
Open the device0
Load the xclbin ./build_dir.sw_emu.xilinx_u280_xdma_201920_3/vadd.xclbin
Allocate Buffer in Global Memory
synchronize input buffer data to device global memory
Execution of the kernel
Get the output data from the device
TEST PASSED

Can you please share more details which Vitis & XRT Version used.

Thanks,
Vishnu

uday610, 2022-01-13:
closed as there was no reply!

closed 2022-01-13 by uday610