Xilinx/XRT #6021: XRT and WSL2

danchitnis, 2021-11-24:
Hello, I am trying to install the Alveo U250 software package for development only (I don't have the card) on WSL2 in Win11 Insider preview. Do I need to install the XRT module for development only? In that case, when I install xrt using `sudo apt install xrt...`, the installer goes into an infinite loop:

```bash
xocl.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-74-generic/updates/dkms/

xclmgmt.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.0-74-generic/updates/dkms/

depmod...

DKMS: install completed.
```

I believe xocl and xclmgmt are Xilinx's kernel drivers for PCIe.

sonals, 2021-11-30:
WSL2 is not a supported platform so this will require your help to resolve. 

Can you dig through to see what is the causing the infinite loop -- the log files left behind by the installer may be helpful here.

danchitnis, 2021-11-30:
WSL2 doesn't have `systemd` (yet). So when `xrt` is trying to install a service and start it, it fails and goes to an infinite loop. This will cause a problem when installing inside containers since in the majority of scenarios there is no root access for `systemd`. The service startup doesn't seem essential for development purposes, since my Vitis seem to function correctly. 

Where can I find the logs for the xrt installer?

keryell, 2021-12-04:
Are you compiling from the sources?
This comes probably from https://github.com/Xilinx/XRT/blob/master/build/debian/xrt-xocl-dkms.postinst
You can instrument this file, rerun it after you stopped the failing install.
I find replacing `#! /bin/sh` by `#! /bin/sh -vx` useful to debug shell scripts.
PR welcome. :-)
I am curious about whether GitHub Action allows to try this with Windows + WSL2...

danchitnis, 2021-12-06:
@keryell I used [this link](https://www.xilinx.com/bin/public/openDownload?filename=xrt_202120.2.12.427_20.04-amd64-xrt.deb) in the Alveo U250 page. It may differ with the [source](https://github.com/Xilinx/XRT/blob/master/build/debian/xrt-xocl-dkms.postinst) because the messages seem to be different. 

The crucial question is whether the development (i.e. SW and HW emulation) works without installing the `xrt` package? WSL doesn't have `systemctl`, so these services will not run anyway. 

> I am curious about whether GitHub Action allows to try this with Windows + WSL2...

Yes, it should be possible. Currently, WSL is very similar to a rootless container, so a docker container in Github action can replicate the results of installing on WSL. I will try to create a minimal repository to demonstrate the `xrt` problem.

keryell, 2021-12-07:
All this is unsupported platform, so you will not be able to install the official XRT package.
You can always deconstruct the `.deb`, edit the control files and reassemble it so it skip or at least does not fail on the `systemctl` part.
Otherwise another way to get XRT working might be to compile it to the platform but skipping the device-driver install because anyway it cannot work yet on WSL2.

danchitnis, 2021-12-08:
correct, I think the script can check if `systemctl` returns without error and then exits the postinst process. I will create a pull requires on this later.

I have created a [repository](https://github.com/danchitnis/xrt-dev-only) to test the xrt installations. It doesn't have the infinite loop which I encountered in my WSL install. Eventually, I will expand it to a full installation (excluding Vitis due to size) which can be freshly built and pulled directly from a container repository.

keryell, 2021-12-08:
Thank you for working on this!
@lforg37: I guess you had a similar issue when trying to make a Docker image with Vitis + XRT? I guess it was not failing that hard?