Xilinx/XRT #6027: Strange XILINX_VITIS default paths

keryell, 2021-11-29:
While trying to solve how to compile ERT as an outsider, I ran into some strange configurations found in the source tree:
- around https://github.com/Xilinx/XRT/blob/master/build/build.sh#L227 :
  ```bash
  # we pick microblaze toolchain from Vitis install
  if [[ -z ${XILINX_VITIS:+x} ]]; then
    export XILINX_VITIS=/proj/xbuilds/2019.2_released/installs/lin64/Vitis/2019.2
  fi
  ```
- around https://github.com/Xilinx/XRT/blob/master/build/run.sh#L10 :
  ```bash
  # Set to location of your preferred Vitis install
  vitis=/proj/xbuilds/2021.1_released/installs/lin64/Vitis/2021.1
  ```
  it is *very unlikely* it is always the preferred Vitis install of the reader. :-)

While this sounds like a good idea to peak some default, at the end it might lead to very subtle errors and rapid code churning, so it might be preferable to just remove this random default picks.

Probably https://github.com/Xilinx/XRT/blob/master/src/runtime_src/doc/toc/test.rst would require some update on the versions and tools too.

stsoe, 2021-12-02:
@keryell .  Yeah run.sh is not really meant for public consumption and the defaults value are no good.  I agree.
I am not sure what I can do about the default in build.sh other than update it to a more recent version.

keryell, 2021-12-04:
I would say to just remove these default values, so it does not gives the false impression it works everywhere while it just works by chance and we do not understand why it does not work when it is tried outside of Xilinx.

stsoe, 2021-12-10:
I can't remove the default from build.sh as it would require everyone building XRT to have XILINX_VITIS set.  Our developers don't do that and build.sh is used primarily by our developers.   I think I will warn if XILINX_VITIS is not set **and** the default path is not available.

keryell, 2021-12-11:
It seems easy to educate your developers to have `XILINX_VITIS` set with some configuration file rather to pick an old `/proj/xbuilds/2021.1_released/installs/lin64/Vitis/2021.1` or obsolete `/proj/xbuilds/2019.2_released/installs/lin64/Vitis/2019.2` and banging one's head on why it works on some machines and not on another ones.
So, at the end you think that our developers are not motivated enough by setting `XILINX_VITIS` while you assume our super motivated customers will be Adaptable enough to build XRT after setting `XILINX_VITIS`? ;-) @j-stephan 
That said, I will not die for since as I have understood the problem which happens only inside Xilinx and documented it in my SYCL XRT build recipe.

stsoe, 2021-12-11:
Yes, I have much more confidence that you and motivated customers understand how to build XRT than I do educating the vast number of internal customers :-)  Maybe the 'how-to-build' section needs updating to be more specific, I haven't checked.

PR #6114 (2022-01-11): Issue #6027 Update Vitis default paths in scripts
#### Problem solved by the commit
Update default VITIS paths in build scripts

#### Bug / issue (if any) fixed, which PR introduced the bug, how it was discovered
Issues #6027 raises concerns about magically defaulting Vitis paths in build.sh.

#### How problem was solved, alternative solutions (if any) and why they were rejected
Defaulting the paths will not change, but they are updated to more recent versions and warning is issued if path cannot be found. Removing the paths and relying on internal users managing the setting of XILINX_VITIS is [... cut]
commit c95d93cd (2022-01-11): Issue #6027 Update Vitis default paths in scripts (#6114)

Issues #6027 raises concerns about magically defaulting Vitis paths in
build.sh.

Defaulting the paths will not change, but they are updated to more
recent versions and warning is issued if path cannot be found.
Removing the paths and relying on internal users managing the setting
of XILINX_VITIS is not feasible.  The scripts are primarily for
convienience of internal users.

Risk is minimal resulting from path update in build.sh and
preservation of XILINX_XRT in run.sh.

This PR closes #6027.
closed 2022-01-11 by stsoe