Xilinx/XRT #6033: system freeze while installing XRT

whyaysd, 2021-12-01:
Hi,

I am installing XRT for vck5000. However, the system always hang when I install the xrt package. I have try this in two methods, both of them freezed.
1. using the install.sh script provided in Visit-AI
2. build XRT from source code of 2021.2.

My OS is ubuntu 20.04 kernel 5.4.0

I need your help to solve this problem. 

Thank you!

xdavidz, 2021-12-02:
@whyaysd please provide more info:
1) What shell pkg you have installed or you don't have shell installed?
2) What sc version you have installed or you don't have sc pkg installed?
3) Where is the pkg of XRT in Vitis-AI?
4) Could you please list your vck5000 device id by "lspci -vd 10ee:"?

whyaysd, 2021-12-02:
1. bash?
2. Sorry, I don't understand the sc pkg
3. https://github.com/Xilinx/XRT/releases/tag/202120.2.12.427
4. the output is
```
01:00.0 Memory controller: Xilinx Corporation Device 5044
	Subsystem: Xilinx Corporation Device 000e
	Flags: fast devsel, IRQ 255
	Memory at a0000000 (64-bit, prefetchable) [disabled] [size=128M]
	Memory at a8020000 (64-bit, prefetchable) [disabled] [size=128K]
	Capabilities: [40] Power Management version 3
	Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [60] MSI-X: Enable- Count=32 Masked-
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [180] Alternative Routing-ID Interpretation (ARI)
	Capabilities: [1c0] Secondary PCI Express
	Capabilities: [3a0] Data Link Feature <?>
	Capabilities: [600] Vendor Specific Information: ID=0020 Rev=0 Len=010 <?>

01:00.1 Memory controller: Xilinx Corporation Device 5045
	Subsystem: Xilinx Corporation Device 000e
	Flags: fast devsel, IRQ 255
	Memory at a8000000 (64-bit, prefetchable) [disabled] [size=128K]
	Memory at 90000000 (64-bit, prefetchable) [disabled] [size=256M]
	Capabilities: [40] Power Management version 3
	Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [60] MSI-X: Enable- Count=32 Masked-
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [180] Alternative Routing-ID Interpretation (ARI)
	Capabilities: [600] Vendor Specific Information: ID=0020 Rev=0 Len=010 <?>
```
5.there are some log in dmesg saying: xclmgmt: module verification failed: signature and/or required key missing - tainting kernel

xdavidz, 2021-12-02:
1. Not the bash shell. The pkg for the vck5000 card, should be something like "xilinx-vck5000-gen3x16-xdma-base_1-3378008_all.deb". If you never installed this, then it is not related to vck5000 board itself. You can verify by removing the card and re-install XRT. You might see the same error.
2. I am pretty sure you haven't installed No.1, otherwise it will request you to install the sc-pkg. Satellite-Controller pkg.
3. 2021.2 should work.
4. This is a es-1 card, please make sure use es-1 xdma pkg(we call it shell for the card).
5. I am thinking it is due to your ubuntu build environment and not the vck5000 card. You can remove or disable the pcie slot of the vck5000 card, and re-install XRT. Then see if there are any difference.

whyaysd, 2021-12-02:
thank you very much.

As you recommended,  I have tried to power off my computer and remove the power cable of vck5000 and then re-install XRT, but the problem is all the same.

whyaysd, 2021-12-02:
There are some more details:
1. First of all, I install a new ubuntu 20.04.1 LTS OS.
2. I install git vim using apt
3. download the XRT source code release
4. install dependent software using the script provided in XRT source code
5. compiling
6. shutdown my computer, plug in the VCK5000, and restart my computer
7. using lspci to check my vck5000 is conected
8. using apt to install the compiled xrt deb package and this program will hang after it print those messages:
```
.....
DKMS: install completed.
Finished DKMS common.postinst
Loading new XRT Linux kernel modules
```
Then, the computer is frozen, I can not use my keyboard/mouse, and my screen freeze.
I use the timer in my iPhone to measure the freezing time,  which is more than 10 minutes and it seems to last forever.

I have tried many methods to work around it, but all of those end up failing.

I have been stuck in this problem for a couple of days and tried it again and again.

However, I can not expect I could solve this problem. Maybe give-up is a clever choice.

whyaysd, 2021-12-04:
Hi,

I have tried it for other days. And I can update some information regarding your questions.

And I have installed the shell package, but I found that I can't finish the `xbutil validate`. When I execute this, my computer will freeze, leaving the following message:
```
/opt/xilinx/xrt/bin/xbutil validate
INFO: Found 1 cards

INFO: Validating card[0]: xilinx_vck5000-es1_gen3x16_base_2
INFO: == Starting AUX power connector check:
AUX power connector not available. Skipping validation
INFO: == AUX power connector check SKIPPED
INFO: == Starting Power warning check:
INFO: == Power warning check PASSED
INFO: == Starting PCIE link check:
INFO: == PCIE link check PASSED
INFO: == Starting SC firmware version check:
INFO: == SC firmware version check PASSED
INFO: == Starting verify kernel test:
```

Even if i do not use vck5000, I mean, do not do anything about vck 5000, my computer will also freeze.

> 1. What shell pkg you have installed or you don't have shell installed?
My board is ES board, so I follow the security site's guide and I install the shell from this page: https://github.com/Xilinx/Vitis-AI/tree/master/setup/vck5000

> 2. What sc version you have installed or you don't have sc pkg installed?
```
 apt list --installed | grep xilinx

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

xilinx-sc-fw-vck5000/now 4.4.6-2.e1f5e26 all [installed,local]
xilinx-vck5000-es1-gen3x16-base/now 2-3123623 all [installed,local]
xilinx-vck5000-es1-gen3x16-validate/now 2-3123623 all [installed,local]
```
> 3. Where is the pkg of XRT in Vitis-AI?
I build it from https://github.com/Xilinx/XRT/releases/tag/202120.2.12.427

> 4. Could you please list your vck5000 device id by "lspci -vd 10ee:"?
```
sudo lspci -vd 10ee:
01:00.0 Memory controller: Xilinx Corporation Device 5044
	Subsystem: Xilinx Corporation Device 000e
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Memory at a0000000 (64-bit, prefetchable) [size=128M]
	Memory at a8020000 (64-bit, prefetchable) [size=128K]
	Capabilities: [40] Power Management version 3
	Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [60] MSI-X: Enable- Count=32 Masked-
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [180] Alternative Routing-ID Interpretation (ARI)
	Capabilities: [1c0] Secondary PCI Express
	Capabilities: [3a0] Data Link Feature <?>
	Capabilities: [600] Vendor Specific Information: ID=0020 Rev=0 Len=010 <?>
	Kernel driver in use: xclmgmt
	Kernel modules: xclmgmt

01:00.1 Memory controller: Xilinx Corporation Device 5045
	Subsystem: Xilinx Corporation Device 000e
	Flags: bus master, fast devsel, latency 0, IRQ 17
	Memory at a8000000 (64-bit, prefetchable) [size=128K]
	Memory at 90000000 (64-bit, prefetchable) [size=256M]
	Capabilities: [40] Power Management version 3
	Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [60] MSI-X: Enable+ Count=32 Masked-
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [180] Alternative Routing-ID Interpretation (ARI)
	Capabilities: [600] Vendor Specific Information: ID=0020 Rev=0 Len=010 <?>
	Kernel driver in use: xocl
	Kernel modules: xocl
```

5. Were there log in the dmesg?
[dmesg.0.txt](https://github.com/Xilinx/XRT/files/7654423/dmesg.0.txt)
[dmesg.1.txt](https://github.com/Xilinx/XRT/files/7654424/dmesg.1.txt)

xdavidz, 2021-12-04:
Please adjust your steps from:

1. First of all, I install a new ubuntu 20.04.1 LTS OS.
2. I install git vim using apt
3. download the XRT source code release
4. install dependent software using the script provided in XRT source code
5. compiling
6. shutdown my computer, plug in the VCK5000, and restart my computer
7. using lspci to check my vck5000 is conected
8. using apt to install the compiled xrt deb package and this program will hang after it print those messages:

To

1. First of all, I install a new ubuntu 20.04.1 LTS OS.
2. I install git vim using apt
3. download the XRT source code release
4. install dependent software using the script provided in XRT source code
5. compiling
6. using apt to install the compiled xrt deb package
6. shutdown my computer, plug in the VCK5000, and restart my computer
7. using lspci to check my vck5000 is conected
8. using apt to install sc pkg and vck5000 validate and base pkg
9. using xbmgmt examine and xbutil examine check the output 

Please let me know at which step you got your system stuck, then I can help you from there.

whyaysd, 2021-12-05:
I think the main difference is the order between install xrt and vck5000. If I don't plug in vck5000, installing xrt will not stuck my OS.

If I install xrt with vck5000 installed, I should using kernel 5.8.0-63-generic, or my os will also freeze during the installation.

Regarding to:
> 8. using lspci to check my vck5000 is conected
> 9. using apt to install sc pkg and vck5000 validate and base pkg
> 10. using xbmgmt examine and xbutil examine check the output

The system will definitely stuck if I using xbutil validate. others operation seems ok, but sometimies my system will stuck even if I don't do anything

Besides, I found that `sudo /opt/xilinx/xrt/bin/xbmgmt flash --update` never trigger the update, always gives following message:
```
sudo /opt/xilinx/xrt/bin/xbmgmt flash --update
Card [0000:01:00.0]:
	 Status: shell is up-to-date

Card(s) up-to-date and do not need to be flashed.
```
I don't know if I need to program/revert the flash on VCK5000 ES1 card to factory image by using Vivado via JTAG.

xdavidz, 2021-12-05:
may you please provide more details?
have you done step6? which passed without an issue correct?
in step7, your system won't boot? 

There is no need to re-install XRT in step7.

Please just reproduce the exact steps that I have just asked.

whyaysd, 2021-12-05:
I have done step 6
In step 7 my system boot successfully and I can login successfully, but system will hang for a while or hang immediately If I execute xbutil validate.

I do not re-installed XRT in step

> Please just reproduce the exact steps that I have just asked.
Should I change my kernel version? If I don't It will hang in step 6

xdavidz, 2021-12-05:
It seems your OS system is ok, because step6 passed. meaning you can install XRT without vck5000 card.
Even the step7 passed. meaning you can load driver with vck5000 card.

I think you haven't told me that output of step8 and step9, which is very important for me to narrow down the issue.

Let's hold off "xbutil validte" for now. It is not even in my list till step 10.

If any steps between 1 to 10, please attach the entire dmesg for me to debug.
thanks.

whyaysd, 2021-12-06:
For step 8, I have already post its result before:
```
sudo lspci -vd 10ee:
01:00.0 Memory controller: Xilinx Corporation Device 5044
	Subsystem: Xilinx Corporation Device 000e
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Memory at a0000000 (64-bit, prefetchable) [size=128M]
	Memory at a8020000 (64-bit, prefetchable) [size=128K]
	Capabilities: [40] Power Management version 3
	Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [60] MSI-X: Enable- Count=32 Masked-
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [180] Alternative Routing-ID Interpretation (ARI)
	Capabilities: [1c0] Secondary PCI Express
	Capabilities: [3a0] Data Link Feature <?>
	Capabilities: [600] Vendor Specific Information: ID=0020 Rev=0 Len=010 <?>
	Kernel driver in use: xclmgmt
	Kernel modules: xclmgmt

01:00.1 Memory controller: Xilinx Corporation Device 5045
	Subsystem: Xilinx Corporation Device 000e
	Flags: bus master, fast devsel, latency 0, IRQ 17
	Memory at a8000000 (64-bit, prefetchable) [size=128K]
	Memory at 90000000 (64-bit, prefetchable) [size=256M]
	Capabilities: [40] Power Management version 3
	Capabilities: [48] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [60] MSI-X: Enable+ Count=32 Masked-
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [180] Alternative Routing-ID Interpretation (ARI)
	Capabilities: [600] Vendor Specific Information: ID=0020 Rev=0 Len=010 <?>
	Kernel driver in use: xocl
	Kernel modules: xocl
```

For step 9, I can install those package successfully.

xdavidz, 2021-12-06:
I guess you mean step10 is also good? What are the outputs of xbutil examine and xbmgmt examine?

rahulsingh701, 2021-12-28:
We have received a VCK5000 Card in the xilinx adaptive computing challenge, we are getting this same error.
@whyaysd @xdavidz have you got any working solution for this

xdavidz, 2021-12-28:
@rahulsingh701 please try the exact following steps:

1. First of all, I install a new ubuntu 20.04.1 LTS OS.
2. I install git vim using apt
3. download the XRT source code release
4. install dependent software using the script provided in XRT source code
5. compiling
6. using apt to install the compiled xrt deb package
7. shutdown my computer, plug in the VCK5000, and restart my computer
8. using lspci to check my vck5000 is conected
9. using apt to install sc pkg and vck5000 validate and base pkg
10. using xbmgmt examine and xbutil examine check the output

please do not run xbutil validate just yet, and let me know what you will get at the step 10.

rahulsingh701, 2021-12-29:
@xdavidz we tried until step 7, When I insert the card and try to boot my pc is not starting.
Because of the kernel module installed in the XRT, here is the error image:
[image]
@xdavidz can you help me on this?

ombhilare999, 2021-12-29:
We got some work around @xdavidz .
We installed xrt from master itself as the latest release was failing.

xdavidz, 2021-12-29:
@rahulsingh701 thanks for the info. This error indicates that you've installed a shell pkg (vck5000-xdma etc..) doesn't match the existing shell flashed on to the card. This is a known issue. Do you have jtag usb connected to the card? I can help to revert back to recovery mode and re-flash the card.
@ombhilare999 if I understand it correctly, you can run "xbutil examine" multiple times, however if you warm reboot the machine, the machine will hung?What is the entire output of the "xbutil examine"?

VedantParanjape, 2021-12-29:
> @rahulsingh701 thanks for the info. This error indicates that you've installed a shell pkg (vck5000-xdma etc..) doesn't match the existing shell flashed on to the card. This is a known issue. Do you [...]

Hey @xdavidz I am working with @rahulsingh701 on the same card. What do you mean by JTAG USB ? The micro usb port on the card ?

Also, we weren't able to install anything on the card, can you please help us revert back to recovery mode and reflash.

Regards and Thanks,
Vedant

xdavidz, 2021-12-29:
Are you working for Xilinx? Please slack me.

VedantParanjape, 2021-12-29:
> Are you working for Xilinx? Please slack or teams me.

Hello, No, we got the Engineering sample for a competition by Xilinx. What's your Slack or teams ID ? (I don't think we are on same server)

xdavidz, 2021-12-29:
search davidzha@xlinx.com please

VedantParanjape, 2021-12-29:
> search [davidzha@xlinx.com](mailto:davidzha@xlinx.com) please

I have emailed you.

VedantParanjape, 2021-12-29:
@xdavidz what's the use of the micro usb port on the card ?

xdavidz, 2021-12-29:
@VedantParanjape I think officially I should support you guys here. The micro usb is used for programing the board by Vivado.
I don't know your cards status, but if you have the micro usb connected, then we can check more info on the card. The card is running a petalinux onto it.

VedantParanjape, 2021-12-30:
> @VedantParanjape I think officially I should support you guys here. The micro usb is used for programing the board by Vivado.
> I don't know your cards status, but if you have the micro usb connected, then we can check more info on the card. The card is running a petalinux onto it.

Thanks David for the support. Can you list the commands, I'll run on my host pc to which the micro usb is connected.

VedantParanjape, 2022-01-03:
> @VedantParanjape I think officially I should support you guys here. The micro usb is used for programing the board by Vivado.
> I don't know your cards status, but if you have the micro usb connected, then we can check more info on the card. The card is running a petalinux onto it.

Hello any update?

xdavidz, 2022-01-03:
@VedantParanjape please contact me via email davidzha@xlinx.com

Blatts01, 2022-01-12:
Hi xdavidz,

@xdavidz 
can I contact you to check my vck5000 via console? I can log in to the petalinux system on the card. But my system crash while loading the xocl.ko module. 

Best Regards 
Stefan

Blatts01, 2022-01-12:
Ok, blacklist help booting the system, but it crash if i try to load the xocl.ko module, while the vck5000 card is installed

Blatts01, 2022-01-12:
i just send you a slack Invitation, lunzt_es_euch_rein ,the david hasselhoff guy :)

xdavidz, 2022-01-12:
> i just send you a slack Invitation, lunzt_es_euch_rein ,the david hasselhoff guy :)

seems I cannot accept invitation from request outside the company. let's debug the issue here.
1) could you please balcklist xocl and xclmgmt when server start.
2) once your system is up, please ONLY load the xclmgmt driver by "modprobe xclmgmt".
3) meanwile, please keep all dmesg by dmesg -w or config it to at system log

Please let me know what exactly you have in the dmesg during modprobe xclmgmt.

Blatts01, 2022-01-12:
Hi xdavidz,

here the modprobe console:

$ sudo modprobe xclmgmt 
[sudo] Passwort für blattmann: 
modprobe: ERROR: could not insert 'xclmgmt': Unknown symbol in module, or unknown parameter (see dmesg)
$ uname -r 
5.8.0-50-generic

dmesg -w output 

[   78.524338] xclmgmt: Unknown symbol fpga_mgr_create (err -2)
[   78.524488] xclmgmt: Unknown symbol fpga_mgr_unregister (err -2)
[   78.524600] xclmgmt: Unknown symbol fpga_mgr_register (err -2)
[   78.524622] xclmgmt: Unknown symbol fpga_mgr_free (err -2)

xdavidz, 2022-01-12:
> Hi xdavidz,
>
> here the modprobe console:
>
> $ sudo modprobe xclmgmt [sudo] Passwort für blattmann: modprobe: ERROR: could not insert 'xclmgmt': Unknown symbol in module, or unknown parameter (see [...]

adjusted steps:
1) could you please balcklist xocl and xclmgmt when server start.
2) once your system is up, please ONLY load the xclmgmt driver by "modprobe fpga_mgr; modprobe xclmgmt".
3) meanwile, please keep all dmesg by dmesg -w or config it to at system log

Blatts01, 2022-01-13:
i cant load fpga_mgr the module is not compiled for the kernel. 
I used the vck5000 install.sh script from vitis-ai. Only the xclmgmt.ko and xocl.ko are compiled while xrt installation

xdavidz, 2022-02-11:
I got update from @VedantParanjape he successfully installed vck5000 on R740. 
If anyone still have issues please try to use dell R740 as the host server.

close this issue for now.

closed 2022-02-11 by xdavidz

frenteD, 2023-07-06:
> 

> i cant load fpga_mgr the module is not compiled for the kernel. I used the vck5000 install.sh script from vitis-ai. Only the xclmgmt.ko and xocl.ko are compiled while xrt installation

hi, I have encountered the same problem, how did you solve it?

xdavidz, 2023-07-06:
> 

Try this "modprobe fpga_mgr; modprobe xclmgmt"