Xilinx/XRT #6053: Speeding up MMIO R/W latency

abejgonzalez, 2021-12-08:
I am porting a project from using the AWS EC2 F1 platform to locally hosted U250s and I noticed a large performance difference between the two platforms when using MMIO R/W requests.

Here are the numbers that I collected (measuring around the `ip::write_register` `ip::read_register` API):

*Vitis/XRT U250 MMIO R/W latencies:*

Read (Mean, Median, Min, Max): 123, 113, 105, 685 usec
Write (Mean, Median, Min, Max): 83, 82, 78, 1514 usec

*Compared to AWS EC2 F1's MMIO R/W latencies:*

Read (Mean, Median, Min, Max): 1704, 1637, 696, 53583 nsec
Write (Mean, Median, Min, Max): 761, 698, 677, 55133 nsec

Is there any way to lower the MMIO R/W latencies on Vitis/XRT (there seems to be a 2 order of magnitude difference between the two platforms)? Unlike a "normal" user which would likely write/read MMIO registers at the beginning of the kernel running, our project continuously read/writes to the kernel using MMIO throughout the kernel's runtime which makes it extremely latency-sensitive. Maybe there is a way to modify the XRT runtime and have a custom-built version of XRT (if so any pointers)?

**Further Context/Links:**
Specific APIs used: https://github.com/firesim/firesim/blob/6df5134c42b75a721d63dac77215a1bd9ab9c1a1/sim/midas/src/main/cc/simif_vitis.cc#L63-L82
Open-source Project (FireSim): https://fires.im/
XRT/Xilinx version: 2020.1

Thanks in advance.

stsoe, 2022-01-04:
Hi @abejgonzalez  Sorry for the delay in responding.  We checked the code internally and noticed that the xrt::ip::write/read_register has a sysfs call overhead per invocation.   This probably explains the difference as the code you pointed to for F1 and Vitis is otherwise identical in terms of reading and writing the bar.  We will work on fixing this in XRT:master branch.

closed 2022-01-05 by stsoe

abejgonzalez, 2022-01-15:
As a followup, I can confirm that the changes did result in a large improvement:

Vitis/XRT U250 MMIO Latencies:
Write: ~2.4usec on average
Read: ~3.5usec on average

For now, AWS F1 is still faster (maybe there are some optimizations that can be done?) but this is much improved and for now is sufficient! Thanks!

PR #6110 (2022-01-05): Issue #6053 Speeding up MMIO R/W latency
#### Problem solved by the commit
Remove repeated sysfs call from xclRegRW

#### Bug / issue (if any) fixed, which PR introduced the bug, how it was discovered
Fixes #6053. The issue identified increased latency of MMIO when using xrt::ip::read/write_register.  Tracked to xclRegRW where sysfs was used at every call to check the size of the IP. 

#### How problem was solved, alternative solutions (if any) and why they were rejected
 Moved sysfs call to the initialization phase for one time check.

#### Risks (if any) associated the changes in the commit
 [... cut]