Xilinx/XRT #6061: Xilinx OpenCL Pipes lead to error: Global program variables are not supported.

Mellich, 2021-12-10:
Hi everyone,

I tried to synthesize and execute an OpenCL design that makes use of [pipes](https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/Reducing-Kernel-to-Kernel-Communication-Latency-in-OpenCL-Kernels) with Vitis 2020.2 and XRT 2.9 for the platform `xilinx_u280_xdma_201920_3`.

Execution of the kernels throws the error `[XRT] ERROR: Global program variables are not supported.`. The very same design worked with Vitis 2019.2 and (I think) XRT 2.3.
In the [XRT Changelog](https://github.com/Xilinx/XRT/blob/master/CHANGELOG.rst) it is mentioned for version 2.9, that the implementation of OpenCL changed to use native XRT APIs. May this be the reason for this changed behavior? Unfortunately, I can not test my design with older XRT versions to verify this or get a working setup again.

Is this a bug or does this mean pipes are not supported by XRT anymore? 

Thanks in advance!

stsoe, 2022-01-03:
@Mellich Looks like our documentation is not quite accurate and that example links should be updated.  We no longer support pipes, the recommended approach is to use hls::stream for kernel-to-kernel streaming.  You will find several such examples under https://github.com/Xilinx/Vitis_Accel_Examples, e.g.  https://github.com/Xilinx/Vitis_Accel_Examples/blob/master/cpp_kernels/dataflow_stream/src/adder.cpp

uday610, 2022-01-05:
We will update ug1393 to remove/deprecate OpenCL kernel development section.

Mellich, 2022-01-06:
@stsoe Thank you for the clarification.