Xilinx/XRT #6096: The question about XRT source code Sync()

pyhundan, 2021-12-23:
Hi,
  When I read XRT source code and the XRT [Test](https://github.com/Xilinx/XRT/blob/master/tests/xrt/02_simple/main.cpp#L63) code I have a question. 

   Firstly, I want to realize the data how to transfer between host and device, and I have read the [website](https://xilinx.github.io/XRT/2020.2/html/xrt_native_apis.html), found that `Sync ()` is used to transfer data. 

​	I read the XRT source code and debug through printk, This function is the last call to found him to be called `src/runtime_src/core/pcie/Linux/shim.cpp` in `int shim: : xclSyncBO ().`

​    But I can't find more at the bottom of the call, I only found about `drm_xocl_sync_bo` data structure, but the IOCTL implementation I read the `src/runtime_src/core/pcie/driver/` has not been seen in Linux code, hoping to give some advice.

Thank,
hundan

pyhundan, 2021-12-28:
I have another question. When I changed the code in XRT, it was like this [code](https://github.com/Xilinx/XRT/blob/master/src/runtime_src/core/pcie/driver/linux/xocl/subdev/xmc.c#L553),I added some information inside xocl_info that I wanted to output。
But I run the `test/xrt/02_sample` and use `dmesg` didn't show the corresponding information in the terminal.
the `dmesg` just show like this:
`[108195.713597] xocl 0000:3b:00.1:  ffff8cdf3a60d0b0 _xocl_drvinst_open: OPEN 1
[108195.713602] [drm] creating scheduler client for pid(21422), ret: 0
[108195.718851] xocl 0000:3b:00.1: xmc.u.18874368 ffff8cdf36467410 xmc_read_from_peer: reading from peer
[108195.718862] xocl 0000:3b:00.1: mailbox.u.9437184 ffff8cdf36461410 mailbox_request: sending request: 10 via HW
[108195.719182] xclmgmt 0000:3b:00.0: mailbox.m.9437184 ffff8cdf37830410 process_request: received request from peer: 10, passed on
[108195.719194] xclmgmt 0000:3b:00.0: xclmgmt_read_subdev_req: req kind 0
[108195.719268] xclmgmt 0000:3b:00.0: mailbox.m.9437184 ffff8cdf37830410 mailbox_post_response: posting response for: 10 via HW
[108195.725217] xocl 0000:3b:00.1: icap.u.22020096 ffff8cdf36463c10 icap_read_from_peer: reading from peer
[108195.725219] xocl 0000:3b:00.1: mailbox.u.9437184 ffff8cdf36461410 mailbox_request: sending request: 10 via HW
[108195.725524] xclmgmt 0000:3b:00.0: mailbox.m.9437184 ffff8cdf37830410 process_request: received request from peer: 10, passed on
[108195.725529] xclmgmt 0000:3b:00.0: xclmgmt_read_subdev_req: req kind 1
[108195.725544] xclmgmt 0000:3b:00.0: clock.m.28311552 ffff8cdf119bc810 clock_get_freq_by_id: freq = 300
[108195.725551] xclmgmt 0000:3b:00.0: clock.m.28311552 ffff8cdf119bc810 clock_get_freq_by_id: freq = 500
[108195.725554] xclmgmt 0000:3b:00.0: clock.m.28311552 ffff8cdf119bc810 clock_get_freq_by_id: freq = 0
[108195.727554] xclmgmt 0000:3b:00.0: clock.m.28311552 ffff8cdf119bc810 clock_get_freq_counter_khz: khz: 299995
[108195.729552] xclmgmt 0000:3b:00.0: clock.m.28311552 ffff8cdf119bc810 clock_get_freq_counter_khz: khz: 499991
[108195.729555] xclmgmt 0000:3b:00.0: clock.m.28311552 ffff8cdf119bc810 clock_get_freq_counter_khz: khz: 0
[108195.729561] xclmgmt 0000:3b:00.0: mailbox.m.9437184 ffff8cdf37830410 mailbox_post_response: posting response for: 10 via HW
[108195.730018] xocl 0000:3b:00.1: icap.u.22020096 ffff8cdf36463c10 icap_cached_ocl_frequency: no cached data for 3
[108195.756954] xocl 0000:3b:00.1:  ffff8cdf3a60d0b0 xclbin_downloaded: xclbin is already downloaded
[108195.756956] xocl 0000:3b:00.1:  ffff8cdf3a60d0b0 xocl_read_axlf_helper: Loaded xclbin ee401cc1-a3eb-412e-8000-ada4c2ad60b5
[108195.757220] xocl 0000:3b:00.1: icap.u.22020096 ffff8cdf36463c10 icap_lock_bitstream: bitstream ee401cc1-a3eb-412e-8000-ada4c2ad60b5 locked, ref=1
[108195.757221] xocl 0000:3b:00.1:  ffff8cdf3a60d0b0 exec_reset: exec_reset(0) cfg(1)
[108195.757224] xocl 0000:3b:00.1: mb_scheduler.u.11534336 ffff8cdf36465410 client_ioctl_ctx: CTX add(ee401cc1-a3eb-412e-8000-ada4c2ad60b5, pid 21422, cu_idx 0xffffffff) = 0, ctx=1
[108195.757286] [drm] command scheduler is already configured for this device
[108195.757301] xocl 0000:3b:00.1: icap.u.22020096 ffff8cdf36463c10 icap_unlock_bitstream: bitstream ee401cc1-a3eb-412e-8000-ada4c2ad60b5 unlocked, ref=0
[108195.757303] xocl 0000:3b:00.1:  ffff8cdf3a60d0b0 exec_stop: exec_stop(00000000cc555848)
[108195.757309] xocl 0000:3b:00.1: mb_scheduler.u.11534336 ffff8cdf36465410 client_ioctl_ctx: CTX del(ee401cc1-a3eb-412e-8000-ada4c2ad60b5, pid 21422, cu_idx 0xffffffff) = 0, ctx=0
[108195.758988] xocl 0000:3b:00.1: icap.u.22020096 ffff8cdf36463c10 icap_lock_bitstream: bitstream ee401cc1-a3eb-412e-8000-ada4c2ad60b5 locked, ref=1
[108195.758989] xocl 0000:3b:00.1:  ffff8cdf3a60d0b0 exec_reset: exec_reset(0) cfg(1)
[108195.758991] xocl 0000:3b:00.1: mb_scheduler.u.11534336 ffff8cdf36465410 client_ioctl_ctx: CTX add(ee401cc1-a3eb-412e-8000-ada4c2ad60b5, pid 21422, cu_idx 0xffffffff) = 0, ctx=1
[108195.759057] xocl 0000:3b:00.1: mb_scheduler.u.11534336 ffff8cdf36465410 client_ioctl_ctx: CTX add(ee401cc1-a3eb-412e-8000-ada4c2ad60b5, pid 21422, cu_idx 0x0) = 0, ctx=2
[108195.759626] xocl 0000:3b:00.1: mb_scheduler.u.11534336 ffff8cdf36465410 client_ioctl_ctx: CTX del(ee401cc1-a3eb-412e-8000-ada4c2ad60b5, pid 21422, cu_idx 0xffffffff) = 0, ctx=1
[108195.759632] xocl 0000:3b:00.1: icap.u.22020096 ffff8cdf36463c10 icap_unlock_bitstream: bitstream ee401cc1-a3eb-412e-8000-ada4c2ad60b5 unlocked, ref=0
[108195.759633] xocl 0000:3b:00.1:  ffff8cdf3a60d0b0 exec_stop: exec_stop(00000000cc555848)
[108195.759639] xocl 0000:3b:00.1: mb_scheduler.u.11534336 ffff8cdf36465410 client_ioctl_ctx: CTX del(ee401cc1-a3eb-412e-8000-ada4c2ad60b5, pid 21422, cu_idx 0x0) = 0, ctx=0
[108195.799173] [drm] client exits pid(21422)
[108195.799181] xocl 0000:3b:00.1:  ffff8cdf3a60d0b0 xocl_drvinst_close: CLOSE 2
[108195.799186] xocl 0000:3b:00.1:  ffff8cdf3a60d0b0 xocl_drvinst_close: NOTIFY 0000000092d27034`

stsoe, 2022-01-03:
@pyhundan Yeah it is not exactly obvious to find the ioctl entry point called by xclSyncBO.  You will want to look at xocl_bo.c in particular xocl_sync_bo_ioctl()

pyhundan, 2022-01-05:
Hi, @stsoe
     I looked at this before, but I saw the [BO_ENTER](https://github.com/Xilinx/XRT/blob/master/src/runtime_src/core/pcie/driver/linux/xocl/userpf/xocl_bo.c#L812) in the code, and I looked at the definition and I actually used `printk` as a kernel function, but I didn't find any messages in `dmesg`, so I added some `printk` in this code, and even though I recompiled it, But I still ran the test code using `dmesg` and didn't see the corresponding output.
   

    My xrt source code version is 202020.2.8.743.
    I downloaded and install `xrt_202020.2.8.743_16.04-amd64-xrt.deb ` from the official website when I was configuring the environment. 
    But I  use the `source ../XRT - 202020.2.8.743 / build/Release/opt/xilinx XRT/setup. sh` when I want to run the program.

stsoe, 2022-01-05:
@pyhundan I assume you are inserting the modified kernel modules, or that you are installing the .deb that has your changes?

pyhundan, 2022-01-11:
@stsoe Thanks for your reminder. I packaged my compiled XRT and reinstalled it to solve my doubts.

closed 2022-01-11 by pyhundan