Xilinx/XRT #6139: XRT hangs on clCreateProgramWithBinary in hw emulation mode

boxerab, 2022-01-12:
My system:

OS: Ubuntu Hirsute 21.04
Vitis 2021.2
XRT built from source in Debug mode, using 2021.2 branch
Emulation target : zcu 102 base

I have generated the emconfig.json file, placed in binary directory, also set 

`export  XCL_EMULATION_MODE=hw_emu`

When I run my application and it runs `clCreateProgramWithBinary`,
XRT hangs in the while loop below:

```
void hostMemAccessThread(xclhwemhal2::HwEmShim* inst) {
	inst->set_mHostMemAccessThreadStarted(true);
    auto mq2h_helper_ptr = std::make_unique<Q2H_helper>(inst);
    bool sock_ret = false;
    int count = 0;
    while(inst->get_mHostMemAccessThreadStarted() && !sock_ret && count < 71){
        sock_ret = mq2h_helper_ptr->connect_sock();
        count++;
    }
```

It is trying to create a UNIX socket, which never succeeds.  

Any suggestions ?

boxerab, 2022-01-13:
More info: I built the `vadd` demo kernel for hw_emu target, and I get the hang on this binary.
I am attaching the binary here, in case it helps.

[vadd_hw_emu.zip](https://github.com/Xilinx/XRT/files/7865096/vadd_hw_emu.zip)

boxerab, 2022-01-20:
@stsoe @venkatp-xilinx @akasat have you guys been able to reproduce this issue ?

akasat, 2022-01-20:
HI @boxerab  - I see that you have compiled XRT of your own. Can you please confirm if you are able to use pre-built XRT that comes with the Petalinux image for zcu102 base platform. Are you seeing any issues there? Have you made any modifications to the hw emulation driver, and if so can you please share those details as well.

If there is any error message that you see, that will be helpful to know as well.

boxerab, 2022-01-20:
@akasat thanks - can you point to where I can download this version of XRT for Ubuntu Hirsute ?

akasat, 2022-01-21:
Couple of clarifications:
1. I am assuming you are using Ubuntu as your development machine, and not running on the embedded PS.
2. If targeting zcu102 or other embedded platforms, the embedded linux needs to be booted on QEMU. Please see github examples here for embedded platforms:  https://github.com/Xilinx/Vitis_Accel_Examples/tree/master/hello_world. When you run Makefile, you will need to tell is that HOST_ARCH is aarch64, and also point to petalinux image using EDGE_COMMON_SW.

You can download pre-built petalinux image from here: 
https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-platforms.html
Section "Common Images for Embedded Vitis Platforms"

boxerab, 2022-01-21:
@akasat thanks, I have now set up petalinux under qemu and it boots. 
Unfortunately, the disk space is only 300 MB for the zcu 102 image - it is possible to "resize" the image
so that there is more space available ?

boxerab, 2022-01-21:
@akasat thanks. Are you saying that hw emulation will not work on an x86 host  ?

boxerab, 2022-01-24:
So, I have switched to an 18.04 Ubuntu installation, where I installed the latest XRT deb package from the Xilinx site.
I still get the exact same call stack when the program hangs:

```
__GI___select() at select.c:41 0x7ffff6b44e1f	
unix_socket::start_server() at 0x7fffd08ddf38	
unix_socket::unix_socket() at 0x7fffd08de3f5	
xclhwemhal2::HwEmShim::xclLoadBitstreamWorker() at 0x7fffd08a481e	
xclhwemhal2::HwEmShim::xclLoadXclBin() at 0x7fffd08a994c	
xclLoadXclBin() at 0x7fffd088373a	
xrt_core::shim<xrt_core::device_pcie>::load_axlf() at 0x7fffd087969b	
xrt_core::device::load_xclbin() at 0x7fffd1737aa6	
xrt::device::load_xclbin() at 0x7fffd17a9db4	
xrt_xocl::hal2::device::loadXclBin() at 0x7fffd1a16eab	
xocl::device::load_program() at 0x7fffd1ce66da	
clCreateProgramWithBinary() at 0x7fffd1c88b17	
```

boxerab, 2022-01-25:
@stsoe @venkatp-xilinx @akasat this does look like a bug to me. It would be useful to know if I should expect hw emulation to work on x86 when targeting the zcu 102 - didn't see this in the documentation. Thanks!

randyh62, 2022-01-25:
Hardware emulation runs on the X86 processor in the QEMU environment, as described here:
https://docs.xilinx.com/r/en-US/ug1393-vitis-application-acceleration/Running-Emulation

This process is supported for the zcu102 platform, as demonstrated in this tutorial:
https://github.com/Xilinx/Vitis-Tutorials/tree/2021.2/Getting_Started/Vitis

Can you use this to test your system and post your results?

boxerab, 2022-01-25:
@randyh62 thanks. One question - on the actual zynq hardware, the boot.bin specifies a single xclbin file that can be loaded on the PL, and no other. Is this also true for hardware emulation on QEMU, or can I load any xclbin file ?

boxerab, 2022-01-25:
Alright, I can run the qemu petalinux on my x86 machine using the following steps:

1. `$ mkdir ~/vitis/installs/petalinux/2021.2`
1. Download [petalinux](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html)
1. run petalinux installer :
`$ ./petalinux-v2021.2-final-installer.run -d ~/vitis/installs/petalinux/2021.2/`
1. On Ubuntu, reconfigure `dash` to `bash`
`$  sudo dpkg-reconfigure dash`
1. add this line to `.bashrc`
`source /home/aaron/vitis/installs/petalinux/2021.2/settings.sh`
1. QEMU + Xilinx [documentation](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/821395464/QEMU+User+Documentation)
1. petalinux [documentation](https://www.xilinx.com/content/dam/xilinx/support/documentation/sw_manuals/xilinx2021_2/ug1144-petalinux-tools-reference-guide.pdf)
1. https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/862912682/Networking+in+QEMU
1. Download [zcu 102 BSP](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html) and move to `~/vitis/bsp/2021.2` folder
1. Create `zcu 102` project
```
$ cd ~ && petalinux-create -t project -s ~/vitis/bsp/2021.2/xilinx-zcu102-v2021.2-final.bsp
```

Now we can boot `qemu` :

```
$ cd ~/xilinx-zcu102-2021.2/
$ petalinux-boot --qemu --prebuilt 3 --qemu-args "-net nic -net nic -net nic -net nic,netdev=eth0 -netdev user,id=eth0,hostfwd=tcp::1114-:22"
```

1. I have built my xclbin with `hw_emu` target set and copied over to petalinux along with my host binary
1. I have copied `xrt.ini` file to petalinux and put in binary folder with xclbin and host binary
1. I have set env variable `export  XCL_EMULATION_MODE=hw_emu`

Now, when I run my host exe, it can load the xclbin and create a program, but hangs when it tries to enqueue
the kernel. Here is the output:

```
root@xilinx-zcu102-2021_2:~/bin# ./my_program 
[ 2060.149419] zocl-drm amba_pl@0:zyxclmm_drm: zocl_create_client: created KDS client for pid(828), ret: 0
[ 2060.150125] zocl-drm amba_pl@0:zyxclmm_drm: zocl_destroy_client: client exits pid(828)
[ 2060.155931] zocl-drm amba_pl@0:zyxclmm_drm: zocl_create_client: created KDS client for pid(828), ret: 0
[ 2060.156392] zocl-drm amba_pl@0:zyxclmm_drm: zocl_destroy_client: client exits pid(828)
[ 2060.157572] zocl-drm amba_pl@0:zyxclmm_drm: zocl_create_client: created KDS client for pid(828), ret: 0
[ 2060.158938] zocl-drm amba_pl@0:zyxclmm_drm: zocl_destroy_client: client exits pid(828)
Platform found : Xilinx
Selected Platform Vendor : Xilinx
[ 2060.161971] zocl-drm amba_pl@0:zyxclmm_drm: zocl_create_client: created KDS client for pid(828), ret: 0
[ 2060.162430] zocl-drm amba_pl@0:zyxclmm_drm: zocl_destroy_client: client exits pid(828)
[ 2060.162897] zocl-drm amba_pl@0:zyxclmm_drm: zocl_create_client: created KDS client for pid(828), ret: 0
[ 2060.163565] zocl-drm amba_pl@0:zyxclmm_drm: zocl_destroy_client: client exits pid(828)
Device 0 : edge
Device ID : 0xaaaafca06c80
[ 2060.164938] zocl-drm amba_pl@0:zyxclmm_drm: zocl_create_client: created KDS client for pid(828), ret: 0
[ 2060.165391] zocl-drm amba_pl@0:zyxclmm_drm: zocl_destroy_client: client exits pid(828)
[ 2060.165831] zocl-drm amba_pl@0:zyxclmm_drm: zocl_create_client: created KDS client for pid(828), ret: 0
Creating program with binary.
[ 2061.014974] [drm] skip kind 29(AIE_RESOURCES) return code: -22
[ 2061.015013] [drm] zocl_xclbin_read_axlf The XCLBIN already loaded
[ 2061.015369] [drm] zocl_xclbin_read_axlf 4b03df95-3fd8-ff5c-9755-14a5ef34ce6b ret: 0
[ 2061.032729] [drm] bitstream 4b03df95-3fd8-ff5c-9755-14a5ef34ce6b locked, ref=1
[ 2061.033173] zocl-drm amba_pl@0:zyxclmm_drm:  ffff0008003be010 kds_add_context: Client pid(828) add context CU(0xffffffff) shared(true)
[ 2061.034176] zocl-drm amba_pl@0:zyxclmm_drm:  ffff0008003be010 kds_del_context: Client pid(828) del context CU(0xffffffff)
[ 2061.034634] [drm] bitstream 4b03df95-3fd8-ff5c-9755-14a5ef34ce6b unlocked, ref=0
[ 2061.048174] [drm] bitstream 4b03df95-3fd8-ff5c-9755-14a5ef34ce6b locked, ref=1
[ 2061.048700] zocl-drm amba_pl@0:zyxclmm_drm:  ffff0008003be010 kds_add_context: Client pid(828) add context CU(0xffffffff) shared(true)
Created program with binary.
Building program
Build Options are :  -I ./ 
[ 2061.077221] zocl-drm amba_pl@0:zyxclmm_drm:  ffff0008003be010 kds_add_context: Client pid(828) add context CU(0xffffffff) shared(true)
[ 2061.086117] zocl-drm amba_pl@0:zyxclmm_drm:  ffff0008003be010 kds_add_context: Client pid(828) add context CU(0x0) shared(true)
Created kernel krnl_vadd
allocate buffer
setting kernel args
creating queue
mapping buffers
setting buffers
unmapping buffers
enqueue kernel
````

akasat, 2022-01-25:
You are down on wrong path. Please see my earlier reply and the hello_world example link. If you are running PS + PL, then you need to use Vitis tools, and not petalinux. You can use Petalinux tool just to build a customized image if needed, but it is recommended to start with a pre-built image (link shared earlier).

boxerab, 2022-01-26:
Thanks @akasat and everyone else on this thread - I was able to get pre-built petalinux image working with my kernel.

Is it possible to hw emu to load a second kernel after boot ? This would really help speed up my dev cycle, as I could create a new kernel and then just scp the binary to a running petalinux instance.

boxerab, 2022-01-29:
thanks, guys, my issue has been resolved. closing.

closed 2022-01-29 by boxerab

MohamedRaizz, 2023-03-30:
> root@xilinx-zcu102-2021_2:~/bin# ./my_program
> [ 2060.149419] zocl-drm amba_pl@0:zyxclmm_drm: zocl_create_client: created KDS client for pid(828), ret: 0
> [ 2060.150125] zocl-drm [...]
_____________________________________________________________________________________________________

Hi sir,
I also got same issue did u solve that ? could u please help me

 
root@Petalinux:~# show_dpu 
[  72.433918] Internal error: synchronous external abort: 96000010 [#1] SMP
[  72.440699] Modules linked in: xt_conntrack xt_MASQUERADE nf_conntrack_netlink xt_addrtype iptable_filter iptable_nat nf_nat br_netfilter dmaproxy(O) al5d(O) al5e(O) allegro(O) brcmfmac brcmutil mali(q
[  72.461585] CPU: 1 PID: 1158 Comm: show_dpu Tainted: G      O   5.10.0-xilinx-v2021.2 #1
[  72.470356] Hardware name: xlnx,zynqmp (DT)
[  72.474525] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[  72.480536] pc : xrt_cu_hls_init+0xc4/0x120 [zocl]
[  72.485321] lr : xrt_cu_hls_init+0x64/0x120 [zocl]
[  72.490094] sp : ffffffc016ad3520
[  72.493392] x29: ffffffc016ad3520 x28: 000000000000004c 
[  72.498696] x27: ffffff8003f20084 x26: ffffff8003f20378 
[  72.503999] x25: ffffff8003c40c10 x24: ffffff80050e5664 
[  72.509303] x23: ffffff8005194000 x22: ffffff80050e5600 
[  72.514606] x21: ffffffc01178d000 x20: ffffff8005194000 
[  72.519910] x19: ffffff8003bc8800 x18: 0000000000000020 
[  72.525213] x17: 0000000000000000 x16: 0000000000000000 
[  72.530517] x15: ffffff8004e0b630 x14: 0000000080120000 
[  72.535820] x13: ffffffc0114f1638 x12: ffffffc01127a3d8 
[  72.541124] x11: ffffff807feff590 x10: 0000000080120000 
[  72.546427] x9 : ffffffc0162b0000 x8 : ffffffc011409000 
[  72.551731] x7 : 0000004069e70000 x6 : 0000000000000000 
[  72.557034] x5 : 00000002ffffffff x4 : 0000000000000005 
[  72.562337] x3 : 0000000000000000 x2 : ffffffc008e0d000 
[  72.567641] x1 : ffffffc0162a0000 x0 : 0000000000000000 
[  72.572945] Call trace:
[  72.575385] xrt_cu_hls_init+0xc4/0x120 [zocl]
[  72.579820] cu_probe+0x250/0x37c [zocl]
[  72.583731] platform_drv_probe+0x54/0xb0
[  72.587729] really_probe+0xe8/0x4b4
[  72.591287] driver_probe_device+0x58/0xc0
[  72.595367] __device_attach_driver+0xa8/0x10c
[  72.599793] bus_for_each_drv+0x78/0xd0
[  72.603613] __device_attach+0xdc/0x17c
[  72.607432] device_initial_probe+0x14/0x20
[  72.611598] bus_probe_device+0x9c/0xa4
[  72.615419] device_add+0x328/0x714
[  72.618899] platform_device_add+0x114/0x234
[  72.623169] subdev_create_cu+0xb0/0x170 [zocl]
[  72.627690] zocl_create_cu+0xb4/0x1a0 [zocl]
[  72.632038] zocl_xclbin_read_axlf+0x950/0xd20 [zocl]
[  72.637081] zocl_read_axlf_ioctl+0x40/0x64 [zocl]
[  72.641858] drm_ioctl_kernel+0xc0/0x110
[  72.645771] drm_ioctl+0x21c/0x480
[  72.649158] __arm64_sys_ioctl+0xb8/0xe0
[  72.653074] el0_svc_common.constprop.0+0x94/0x1c0
[  72.657854] do_el0_svc+0x44/0xb0
[  72.661155] el0_svc+0x14/0x20
[  72.664200] el0_sync_handler+0x1a4/0x1b0
[  72.668201] el0_sync+0x174/0x180
[  72.671503] Code: b901aa84 f9011282 7100087f 54000120 (b9400021) 
[  72.677586] ---[ end trace 62a6bc6f30adf251 ]---
[  72.683329] zocl-drm amba_pl@0:zyxclmm_drm: zocl_destroy_client: client exits pid(1158)
Segmentation fault