Xilinx/XRT #6152: The memory bank index for edge MPSoC/ embedded Ubuntu / XRT

tommy-ietomi, 2022-01-18:
Now I'm currently experimenting acceleration applications on the newly released Embedded [Ubuntu](https://ubuntu.com/download/xilinx) [/XRT/xcl ](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/2037317633/Getting+Started+with+Certified+Ubuntu+20.04+LTS+for+Xilinx+Devices#Pre-Installed-Packages) and MPSoC(Kria SOM).

With referring [here ](https://xilinx.github.io/XRT/master/html/opencl_extension.html#specify-regular-buffer-location-on-the-device-memory-banks), I'd like to confirm.

Q1)Can we use all below buffer location specification methods for edge?
  -Specify a Buffer location by DDR bank name (legacy)
  -Specify a Buffer location by memory bank index
  -Specify a Buffer location by kernel name and argument index
  -"No explicit bank assignment"
Comparing to Alveo Accelerator Card, are there any points to note?

Q2)When I use "memory bank index", how can I know this bank index?
I thought that the HP port's "index" value obtained by xclbinutil corresponds to this "bank index". 
```
$>xclbinutil --info --input ./myip.xclbin
  ~~~~~~~~~
   Name:         HP3
   Index:        2
   Type:         MEM_DRAM
   Base Address: 0x0
  ~~~~~~~~~
```
However, the "bank index" assignments result was following error. It seems that the correct memory bank is not recognized.
```
[XRT] WARNING: Argument '1' of kernel 'convolutor_dna' is allocated in memory bank 'HPC1'; compute unit 'k1' cannot be used with this argument and is ignored.
```

Q3)Should  "No explicit bank assignment" also work for edge Ubuntu&XRT?
Is the request below important &mandatory, right?
`Note: Explicit bank specification is not required in most of the host code development. The XRT can obtain the bank location for the buffer if the buffer is used for setting the kernel arguments right after the buffer creation, i.e. before any enqueue operation on the buffer.
`

stsoe, 2022-01-18:
Hi @tommy-ietomi.  
Q1) yes, all methods of buffer allocation should be available.
Q2) the error says that the connectivity of 'k1' compute unit associated with 'convolutor_dna' is not valid with the buffer being used.  You could check the xclbin in more detail and check the actual connectivity of 'convolutor_dna:k1'.  Start finding the IP_LAYOUT index of the compute unit, then look a CONNECTIVITY section and see which memory bank it maps to.
Q3) yes, auto bank assignment will work, but only the buffer is set as kernel argument prior to first enqueue operation.

tommy-ietomi, 2022-01-19:
Hi @stsoe 
Thank you very much for you reply.
  //Then, I'm very sorry, if possible, would you please edit and remove my IP name "conv**_*na" from your post?
  //This was my mistake, I do apologize for any inconvenience caused.

I'm clear Q1) & Q3).

And about Q3), I dumped CONNECTIVITY section with 
`xclbinutil --dump-section CONNECTIVITY:HTML:connectivity.json --input myip.xclbin`
This is the result. And this "mem_data_index" number is corresponding to "memory bank index", right?
```
{
    "connectivity":
    {
        "m_count": "4",
        "m_connection":
        [
            {
                "arg_index": "1",
                "m_ip_layout_index": "0",
                "mem_data_index": "1"
            },
            {
                "arg_index": "2",
                "m_ip_layout_index": "0",
                "mem_data_index": "1"
            },
            {
                "arg_index": "3",
                "m_ip_layout_index": "0",
                "mem_data_index": "1"
            },
            {
                "arg_index": "4",
                "m_ip_layout_index": "0",
                "mem_data_index": "2"
            }
        ]
    }
}
```
So, this case,
arg1 -> "memory bank=1, 
arg2 -> "memory bank=1, 
arg3 -> "memory bank=1, 
arg4 -> "memory bank=2

stsoe, 2022-01-19:
I understand that you cannot upload your IP meta data directly.  Can you maybe obfuscate the names you want to keep hidden?

tommy-ietomi, 2022-01-20:
@stsoe 
>I understand that you cannot upload your IP meta data directly. 

Thank you very much for your concern.
And exactly, it's difficult to submit all current my .xclbin metadata.
I will experiment with a small design and then report here.

Anyway, what I want to know is how to get the "memory bank index" for specifying a Buffer location.
So, a little more detail on your answer below.
And now I believe "mem_data_index" of "CONNECTIVITY section" in .xclbin is corresponding to "memory bank index" of buffer location.

>You could check the xclbin in more detail and check the actual connectivity of 'zzz:k1'. Start finding the IP_LAYOUT index of the compute unit, then look a CONNECTIVITY section and see which memory bank it maps to.

stsoe, 2022-01-20:
Ok.  Let me try explain what you should do.
Dump these sections of the xclbin
```
% xclbinutil --dump-section EMBEDDED_METADATA:raw:.x.xml --input your_xclbin
% xclbinutil --dump-section CONNECTIVITY:json:c.json --input your_xclbin
% xclbinutil --dump-section MEM_TOPOLOGY::json:m.json --input your_xclbin
% xclbinutil --dump-section IP_LAYOUT::json:i.json --input your_xclbin
```

1. Locate the `zzz:k1` IP in the ip_layout section (`i.json)`.  Record its index counting from 0 in the array, say this is `k1_idx`.
2. In the connectivity section (`c.json`) find the entry where `m_ip_layout_index` is same as `k1_idx` and `arg_index` is same as the argument you are setting (1).  Record the corresponding `em_data_index`, say this is `k1_midx`.
3. Now in the mem_topology section (`m.json`) find the entry with index `k1_midx` and match this up with the error message you originally posted.

stsoe, 2022-04-13:
Closing this issue.  I hope the questions were answered.

closed 2022-04-13 by stsoe