Xilinx/XRT #6202: Error: cannot get kernel start time !  with v2021.2

doonny, 2022-01-29:
I am upgrading my project from 2020.1 to 2021.2, and the profiling function does not seems stable in the new version.
I have a group of kernels that will be launched  for multiple iterations during execution, and in some iteration , the following error shows up:
``
[XRT] ERROR: Profiling info not available, make sure profiling is enabled
Error: cannot get kernel start time ![XRT] ERROR: Profiling info not available, make sure profiling is enabled
Error: cannot get kernel end time !
``
Below is the full log:
``
Platform: Xilinx
Totally 1 device(s) are found
  Using Device 0: edge
Device OpenCL Version: OpenCL 1.0
Device Max Compute Units: 0
Device Max WorkGroup Size: -1
Device Max WorkItem Size: -1
Device Global Memory Size: 4096 MBytes
Device Local Memory Size: 16 KBytes
Device Max Clock Freq: 100 Mhz

Warnning: layer-23 requires padding zero-value feature maps for give param LANE_NUM=16
      original num of feature maps is 20, new value is 32

Warnning: layer-24 requires padding zero-value feature maps for give param LANE_NUM=16
      original num of feature maps is 105, new value is 112
Loading weights from:./data/yolov2_voc/weights.data

50645053 total weights read
Loading reference file from: ./data/yolov2_voc/conv24_conv.data
17745 total output reference read

Loading image file from: ./data/yolov2_voc/image.data

519168 bytes image data read from binary files

----------Executing Layer 1: (conv 1)----------

Launching single work-item kernel memRd

Launching single work-item kernel parmRd

Launching single work-item kernel Conv

Launching single work-item kernel ConvCalc

Launching single work-item kernel MemWr

Waiting for memWr event

Launching single work-item kernel Pool

----------Executing Layer 2: (conv 2)----------

Launching single work-item kernel memRd

Launching single work-item kernel parmRd

Launching single work-item kernel Conv

Launching single work-item kernel ConvCalc

Launching single work-item kernel MemWr

Waiting for memWr event

Launching single work-item kernel Pool

----------Executing Layer 3: (conv 3)----------

Launching single work-item kernel memRd
[... 355 lines cut ...]
                  --connectivity.stream_connect localBuffer_1.fp_wr_done_ch:coreConv_1.fp_wr_done_ch:16
                  --connectivity.stream_connect localBuffer_1.wt_wr_done_ch:coreConv_1.wt_wr_done_ch:16
                  --debug
                  --define YOLOV2
                  --hls.clock 300000000:memRead,parmRead,localBuffer,coreConv,convCalc,memWrite,maxPool,reorg
                  --input_files tmp.hw/memRead.xo
                  --input_files tmp.hw/parmRead.xo
                  --input_files tmp.hw/localBuffer.xo
                  --input_files tmp.hw/coreConv.xo
                  --input_files tmp.hw/convCalc.xo
                  --input_files tmp.hw/memWrite.xo
                  --input_files tmp.hw/maxPool.xo
                  --input_files tmp.hw/reorg.xo
                  --link
                  --optimize 0
                  --output build.hw/conv.xclbin
                  --platform xilinx_zcu102_base_dfx_202120_1
                  --profile.data all:all:all
                  --profile.stall all:all:all
                  --report_level 0
                  --save-temps
                  --target hw
                  --temp_dir ./build.hw
==============================================================================
```

jvillarre, 2022-02-14:
The error message "Profiling info is not available, make sure profiling is enabled" is generated from the OpenCL function clGetEventProfilingInfo and requires that the associated command queue of the event was created with the CL_QUEUE_PROFILING_ENABLE property set.  As a shortcut, the "profile=true" option in the xrt.ini file enables this property on all OpenCL command queues.  In 2020.1, the "profile=true" flag also turned on a lot of our specific profiling implementation, such as generating a summary file and reading counters from IP added during compilation.  In 2021.2, we have kept the "profile=true" shortcut for adding the property, but pulled out all of our specific profiling implementation to their own set of flags so we aren't using one flag to turn on both OpenCL level constructs and the summary and trace information our libraries are creating.

For this design, if you are using the OpenCL function clGetEventProfilingInfo, please enable profiling on the command queues by either setting the property explicitly when you create them or by adding "profile=true" to the xrt.ini file.

doonny, 2022-03-09:
@jvillarre Thx for the reply.  It might be good to add these information in the user guide in a future version.