Xilinx/XRT #6239: Incorrect object owning in OpenCL stack between kernel and context

keryell, 2022-02-07:
It looks like when a `cl_context` is released before a `cl_kernel` depending on that context is released, at `cl_kernel` release time 
an exception is thrown with `No context manager for device` message, which seems to happen in:
```
./runtime_src/core/common/api/context_mgr.cpp:131:  throw std::runtime_error("No context manager for device");
./runtime_src/core/common/api/context_mgr.cpp:142:  throw std::runtime_error("No context manager for device");
```
Problem encountered by @lforg37.
@stsoe I have looked quickly on how dependencies are handled there but I am lost.

stsoe, 2022-02-07:
@keryell You are down the wrong path.   The code you are looking at has nothing to do with cl_context or cl_kernel or how these object ownerships are managed.

The code is managing contexts on compute units, e.g. managing calls to xclOpenContext and xclCloseContext.    Before a kernel can use a kernel compute unit, it (the compute unit) must be acquired by the calling process.  The context on a compute unit is acquired as part of creating a kernel object.

The exception is an internal error, could be a bug.  But there is not enough background in this issue to determine why the exception is thrown; so please update.

lforg37, 2022-02-09:
Hello,

Here is the call stack from the debugger:
```
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument

Thread 1 "single_task_vec" received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737342416704) at pthread_kill.c:44
44      pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737342416704) at pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737342416704) at pthread_kill.c:80
#2  __GI___pthread_kill (threadid=140737342416704, signo=signo@entry=6) at pthread_kill.c:91
#3  0x00007ffff7733476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff77197b7 in __GI_abort () at abort.c:79
#5  0x00007ffff7e48a31 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff7e544ec in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff7e534b9 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff7e53c3a in __gxx_personality_v0 () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007ffff7cba484 in ?? () from /lib/x86_64-linux-gnu/libgcc_s.so.1
#10 0x00007ffff7cbaebe in _Unwind_Resume () from /lib/x86_64-linux-gnu/libgcc_s.so.1
#11 0x00007ffff630981a in xrt_core::context_mgr::close_context (device=0x2e7a800, uuid=..., cuidx=...)
    at /storage/lucf/XRT/src/runtime_src/core/common/api/context_mgr.cpp:140
#12 0x00007ffff6377175 in (anonymous namespace)::ip_context::~ip_context (this=0x2e3aff0, __in_chrg=<optimized out>)
    at /storage/lucf/XRT/src/runtime_src/core/common/api/xrt_kernel.cpp:592
#13 0x00007ffff638f9d2 in std::_Sp_counted_ptr<(anonymous namespace)::ip_context*, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0x2e35c50)
    at /usr/include/c++/11/bits/shared_ptr_base.h:348
#14 0x00007ffff630b022 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x2e35c50) at /usr/include/c++/11/bits/shared_ptr_base.h:168
#15 0x00007ffff630a7e1 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x2e35800, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:705
#16 0x00007ffff6376a68 in std::__shared_ptr<(anonymous namespace)::ip_context, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x2e357f8, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:1154
#17 0x00007ffff6376a84 in std::shared_ptr<(anonymous namespace)::ip_context>::~shared_ptr (this=0x2e357f8, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr.h:122
#18 0x00007ffff6392866 in xrt::kernel_impl::~kernel_impl (this=0x2e35750, __in_chrg=<optimized out>)
    at /storage/lucf/XRT/src/runtime_src/core/common/api/xrt_kernel.cpp:1438
#19 0x00007ffff63a8fc2 in __gnu_cxx::new_allocator<xrt::kernel_impl>::destroy<xrt::kernel_impl> (this=0x2e35750, __p=0x2e35750)
--Type <RET> for more, q to quit, c to continue without paging--
    at /usr/include/c++/11/ext/new_allocator.h:168
#20 0x00007ffff63a8e61 in std::allocator_traits<std::allocator<xrt::kernel_impl> >::destroy<xrt::kernel_impl> (__a=..., __p=0x2e35750)
    at /usr/include/c++/11/bits/alloc_traits.h:531
#21 0x00007ffff63a8b57 in std::_Sp_counted_ptr_inplace<xrt::kernel_impl, std::allocator<xrt::kernel_impl>, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0x2e35740)
    at /usr/include/c++/11/bits/shared_ptr_base.h:528
#22 0x00007ffff67fc048 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x2e35740) at /usr/include/c++/11/bits/shared_ptr_base.h:168
#23 0x00007ffff67fba31 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x2e2fac0, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:705
#24 0x00007ffff68e78a6 in std::__shared_ptr<xrt::kernel_impl, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x2e2fab8, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/shared_ptr_base.h:1154
#25 0x00007ffff68e78c6 in std::shared_ptr<xrt::kernel_impl>::~shared_ptr (this=0x2e2fab8, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr.h:122
#26 0x00007ffff68e78e6 in xrt::kernel::~kernel (this=0x2e2fab8, __in_chrg=<optimized out>) at /storage/lucf/XRT/src/runtime_src/core/include/xrt/xrt_kernel.h:563
#27 0x00007ffff68e8dfe in xocl::kernel::xkr::~xkr (this=0x2e2fab8, __in_chrg=<optimized out>) at /storage/lucf/XRT/src/runtime_src/xocl/core/kernel.h:388
#28 0x00007ffff68f2cfc in std::pair<xocl::device const* const, xocl::kernel::xkr>::~pair (this=0x2e2fab0, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/stl_pair.h:211
#29 0x00007ffff68f2d20 in __gnu_cxx::new_allocator<std::_Rb_tree_node<std::pair<xocl::device const* const, xocl::kernel::xkr> > >::destroy<std::pair<xocl::device const* const, xocl::kernel::xkr> > (this=0x2e9b938, __p=0x2e2fab0) at /usr/include/c++/11/ext/new_allocator.h:168
#30 0x00007ffff68f17b3 in std::allocator_traits<std::allocator<std::_Rb_tree_node<std::pair<xocl::device const* const, xocl::kernel::xkr> > > >::destroy<std::pair<xocl::device const* const, xocl::kernel::xkr> > (__a=..., __p=0x2e2fab0) at /usr/include/c++/11/bits/alloc_traits.h:531
#31 0x00007ffff68ef869 in std::_Rb_tree<xocl::device const*, std::pair<xocl::device const* const, xocl::kernel::xkr>, std::_Select1st<std::pair<xocl::device const* const, xocl::kernel::xkr> >, std::less<xocl::device const*>, std::allocator<std::pair<xocl::device const* const, xocl::kernel::xkr> > >::_M_destroy_node (this=0x2e9b938, 
[... cut ...]
256,0,771.231150,1,API_CALL,24
257,256,771.231848,1,API_CALL,24
258,0,771.251344,1,API_CALL,24
259,258,771.251816,1,API_CALL,24
277,0,772.629281,1,API_CALL,24
278,277,772.630095,1,API_CALL,24
279,0,772.645219,1,API_CALL,24
280,279,772.645579,1,API_CALL,24
282,0,772.679187,1,API_CALL,24
283,282,772.679666,1,API_CALL,24
284,0,772.693245,1,API_CALL,24
285,284,772.693569,1,API_CALL,24
287,0,772.740705,1,API_CALL,24
288,287,772.741417,1,API_CALL,24
289,0,772.752513,1,API_CALL,24
290,289,772.752787,1,API_CALL,24
```

stsoe, 2022-02-09:
@lforg37 Can we get a CR please, with a STR (steps-to-reproduce)?  Thanks!

lforg37, 2022-02-09:
That is CR-1121712. It seems that the bug disappear when no custom xrt.ini is provided, so for the record, here is the problematic xrt.ini:

```
[Runtime]
runtime_log = xrtoutlog
hal_log = hal.log
runtime.verbosity = 5

[Emulation]
print_infos_in_console = true
print_warnings_in_console = true
print_errors_in_console = true

[Debug]
native_xrt_trace = true
app_debug = true
data_trace = coarse
opencl_trace = true
continuous_trace = true
```

stsoe, 2022-02-11:
A CR has been filed against profiling.  XRT's object ownership appears correct.

closed 2022-02-11 by stsoe