Xilinx/XRT #6257: XRT build error

jzymessi, 2022-02-10:
When I run build SH this script, the following problems occur
`XRT/XRT-2020.2/build$ ./build.sh
cmake -DRDI_CCACHE=0 -DCMAKE_BUILD_TYPE=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_TOOLCHAIN_FILE= ../../src
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Host system processor is x86_64
-- Target system processor is x86_64
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'libdrm'
--   Found libdrm, version 2.4.101
-- Looking for DRM - found at /usr 2.4.101
-- Checking for module 'OpenCL'
--   Found OpenCL, version 2.1
-- Looking for OPENCL - found at /usr 2.1 /usr/include
-- Found Git: /usr/bin/git (found version "2.17.1")
-- Looking for GIT - found at /usr/bin/git
-- Found Boost: /usr/include (found version "1.65.1") found components:  system filesystem
-- Could NOT find Curses (missing: CURSES_LIBRARY CURSES_INCLUDE_PATH)
CMake Error at /opt/cmake-3.15.0/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find Curses (missing: CURSES_LIBRARY CURSES_INCLUDE_PATH)
Call Stack (most recent call first):
  /opt/cmake-3.15.0/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /opt/cmake-3.15.0/share/cmake-3.15/Modules/FindCurses.cmake:246 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMake/nativeLnx.cmake:82 (find_package)
  CMakeLists.txt:80 (include)

-- Configuring incomplete, errors occurred!
See also "/home/storagex-server2/zhaoyuan/XRT/XRT-2020.2/build/Debug/CMakeFiles/CMakeOutput.log".

real    0m0.880s
user    0m0.567s
sys     0m0.232s
`
How can I solve this problem?

stsoe, 2022-02-11:
@jzymessi Did you run xrtdeps.sh before attempting to build XRT?

jzymessi, 2022-02-23:
> @jzymessi在尝试构建 XRT 之前，您是否运行过 xrtdeps.sh？

@stsoe Thanks，it has been solved successfully according to your method

closed 2022-02-23 by jzymessi