Xilinx/XRT #6282: Contributing a docker image to build XRT?

mgehre-xlx, 2022-02-16:
I wanted to build XRT locally without installing the required dependencies into my system. Thus I created a small Dockerfile
```
FROM ubuntu:18.04

ENV LANG C.UTF-8
ENV DEBIAN_FRONTEND="noninteractive"

COPY ./src/runtime_src/tools/scripts/xrtdeps.sh .
RUN apt-get update \
    && ./xrtdeps.sh \
    && rm -rf /var/lib/apt/lists/* xrtdeps.sh
```
This is then run via
```
docker build . -t xrt-build
docker run --user $UID -i -t -v $PWD:$PWD -w $PWD/build xrt-build /bin/bash ./build.sh
```
in the top level folder. The resulting build is in `build/Debug` and  `build/Release`  on the host.

Is there interest in adding this to the repository and allowing others to build the same way?

wilderfield, 2022-03-08:
Hi @mgehre-amd,

I know that internally @NaAl-Xlnx leverages Docker for building XRT.

I think your suggestion is generally good for building XRT without disturbing a system’s dependencies. Is that your intent? Or, do you also intend to run applications from within this docker?

I think your Dockerfile could be improved with a build argument to support CentOS / Ubuntu 18 / Ubuntu 20 / Ubuntu X.

My only worry is that this method may not build and/or install the kernel modules successfully. The kernel modules are needed for actual communication with the FPGA.

There is a separate project that maintains Docker images with released XRT builds. Even there users are instructed to install XRT on the machine first (to bring in kernel modules).

 https://github.com/Xilinx/Xilinx_Base_Runtime

mgehre-amd, 2022-03-08:
Hi,
yes, this docker image is only meant for building. It's basically an extended version of `build.sh` which also takes care of installing dependencies in a way that doesn't pollute the users system.
It's true that I didn't use the kernel modules with my XRT build. I only tried sw_emu and hw_emu (via v++).

bryanloz-xilinx, 2022-03-09:
@mgehre-amd your idea inspired me today. Thanks.

Would you mind reviewing: #6380

bryanloz-xilinx, 2022-03-11:
@mgehre-amd @uday610 I think we can close this issue per: https://github.com/Xilinx/XRT/blob/master/src/runtime_src/tools/scripts/build_xrt_with_docker.sh

closed 2022-03-11 by uday610