Xilinx/XRT #6314: The question about reading the XRT source codes

pyhundan, 2022-02-23:
I want to understand how data is transferred from the host memory to the memory on the FPGA. 

And I see this code [here](https://github.com/Xilinx/XRT/blob/master/src/runtime_src/core/pcie/driver/linux/xocl/xocl_drv.h#L802) .

In this code, I found that `DMA_OPS(xdev)->migrate_bo(DMA_DEV(xdev),sgt, to_dev, paddr, chan, len) : 0) `

Maybe migrate_bo migrates bo, and then I go on to find the `migrate_bo` definition.

Such as this [code](https://github.com/Xilinx/XRT/blob/master/src/runtime_src/core/pcie/driver/linux/xocl/xocl_drv.h#L777).

```
/* dma callbacks */
struct xocl_dma_funcs {
	struct xocl_subdev_funcs common_funcs;
	ssize_t (*migrate_bo)(struct platform_device *pdev,
		struct sg_table *sgt, u32 dir, u64 paddr, u32 channel, u64 sz);
	ssize_t (*async_migrate_bo)(struct platform_device *pdev,
		struct sg_table *sgt, u32 dir, u64 paddr, u32 channel, u64 sz,
		void (*callback_fn)(unsigned long cb_hndl, int err), void *tx_ctx);
	int (*ac_chan)(struct platform_device *pdev, u32 dir);
	void (*rel_chan)(struct platform_device *pdev, u32 dir, u32 channel);
	u32 (*get_chan_count)(struct platform_device *pdev);
	u64 (*get_chan_stat)(struct platform_device *pdev, u32 channel,
		u32 write);
	u64 (*get_str_stat)(struct platform_device *pdev, u32 q_idx);
	int (*user_intr_config)(struct platform_device *pdev, u32 intr, bool en);
	int (*user_intr_register)(struct platform_device *pdev, u32 intr,
					irq_handler_t handler, void *arg, int event_fd);
	int (*user_intr_unreg)(struct platform_device *pdev, u32 intr);
};
```

If I understand this code correctly, the `*migrate_bo` is a function pointer. But I can't find the definition. Could you give me some tips? 

Thank,

Hundan.

maxzhen, 2022-03-21:
This is what I can see in the source when I search for "migrate_bo":

C symbol: migrate_bo

  File                         Function        Line
0 linux/xocl/subdev/qdma.c     <global>        827 .migrate_bo = qdma_migrate_bo,
1 linux/xocl/subdev/qdma4.c    <global>        871 .migrate_bo = qdma_migrate_bo,
2 linux/xocl/subdev/xdma.c     <global>        400 .migrate_bo = xdma_migrate_bo,
3 driver/linux/xocl/xocl_drv.h <global>        777 ssize_t (*migrate_bo)(struct platform_device *pdev,
4 driver/linux/xocl/xocl_drv.h xocl_migrate_bo 803 (DMA_CB(xdev, migrate_bo) ? DMA_OPS(xdev)->migrate_bo(DMA_DEV(xdev),

Among them, 0, 1, 2 are three different implementations on specific DMA IPs. 3 is the callback signature. 4 is defining the macro for other parts of the driver to call the callback for performing DMA operations.

Hope this helps...

maxzhen, 2022-04-07:
Closing it since there is no response from submitter for more than a month. I'm assuming that the answer is suffice for now.

closed 2022-04-07 by maxzhen